thonny/thonny #2839: blocking ui when removing esp32 device

kr-g, 2023-06-30:
by disconnecting a mp esp32 device from usb port thonny blocks the ui, or it becomes very very slow responsive.

in addition there is an error window:

```
Traceback (most recent call last):
  File "/usr/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/home/benutzer/repo/thonny/thonny/running.py", line 586, in _cmd_interrupt_with_shortcut
    self._cmd_interrupt()
  File "/home/benutzer/repo/thonny/thonny/running.py", line 545, in _cmd_interrupt
    self._proxy.interrupt()
  File "/home/benutzer/repo/thonny/thonny/plugins/micropython/mp_front.py", line 185, in interrupt
    self._send_msg(ImmediateCommand("interrupt"))
  File "/home/benutzer/repo/thonny/thonny/running.py", line 1146, in _send_msg
    self._proc.stdin.write(serialize_message(msg) + "\n")
BrokenPipeError: [Errno 32] Broken pipe

```

kr-g, 2023-07-01:
it constantly prints into shell panel without entering an error mode

[image]

kr-g, 2023-07-01:
it happens only when removing a device with running code (as it looks like)

kr-g, 2023-07-01:
duplicate of https://github.com/thonny/thonny/issues/2836

aivarannamaa, 2023-07-02:
Thank you for the report! I'll close this and handle the problem under #2836.

closed 2023-07-02 by aivarannamaa