thonny/thonny #2840: thonny.backend: Exception while handling 'upload'

WolleKette, 2023-06-30:
When I use Thonny for MicroPython together with stm32 Devices, I sometimes have the **verry strange issue**, that some files couldn't uploaded to the device.

When the problem occurs, it is always reproducible. You can reset the board (unplug it and plug it back in) and try again as many times as you like. The error will always occur. Then sometimes it is enough to change one letter and then it works. I have not found any logic in it yet. And then what's especially funny is, sometimes I can undo the change and then it works too. Although it had not worked several times before with the same file.

I use Thonny 4.1.1 on MacOS 13.4.1, currently together with a NUCLEO-L476RG Board.

And I have to use a configuration with submit_mode = raw because otherwise, Thonny cannot connect to the device at all. 

> [GenericMicroPython]
> submit_mode = raw
> port = /dev/cu.usbmodem1303
> interrupt_on_connect = True
> sync_time = True
> local_rtc = True
> restart_interpreter_before_run = True

Please dear developers, what should I do?

[backend.log](https://github.com/thonny/thonny/files/11914729/backend.log)

aivarannamaa, 2023-07-02:
Thank you for the report! I'll try to find my Nucleo board and maybe I'm able to reproduce this.

WolleKette, 2023-07-04:
I am quite sure that I have found the problem. The stm boards come with a ST-Link firmware. And yesterday I had the idea if it could be a problem in the firmware. After I updated it, the whole thing seems to run much more stable.

closed 2023-07-06 by WolleKette