thonny/thonny #2841: Install Micropython failes when given local image

arduino12, 2023-06-30:
Thonny 4.1.0
Windows 10
Python 3.10.11
Tk 8.6.13

Run -> Configure Interpreter -> Install or update MicroPython (esptool) ->
≡ (press button with tribar) -> Select local MicroPython image... ->
Install

After pressing "Install" I get immediate error:
```
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\base_flashing_dialog.py", line 434, in _perform_work_and_update_status
    temp_file = self._download_to_temp(download_info)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\esptool_dialog.py", line 259, in _download_to_temp
    return super()._download_to_temp(download_info)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\base_flashing_dialog.py", line 474, in _download_to_temp
    shutil.copyfile(download_url, target_path)
  File "C:\Program Files (x86)\Thonny\lib\shutil.py", line 234, in copyfile
    raise SameFileError("{!r} and {!r} are the same file".format(src, dst))
shutil.SameFileError: 'C:\\Users\\User\\Downloads\\firmware-esp32-LOLIN_S2_MINI.bin' and 'C:\\Users\\User\\Downloads\\firmware-esp32-LOLIN_S2_MINI.bin' are the same file
```

My quickfix was adding an `if` before [line 474](https://github.com/thonny/thonny/blob/master/thonny/plugins/micropython/base_flashing_dialog.py#L479):
```
if download_url != target_path:
    shutil.copyfile(download_url, target_path)
```
It may be better to simply remove this copyfile..

Hope it gets fixed soon so I don't have to manualy change this for my students..

aivarannamaa, 2023-06-30:
Please try Thonny 4.1.1

arduino12, 2023-06-30:
Works! Thanks!

closed 2023-06-30 by arduino12