thonny/thonny #2843: If main.py is loaded in pico, thonny does not communicate with it

DanSkareda, 2023-07-01:
Hello. I'm using the latest version of thonny (4.1.1) and micropython for raspberry pico w (from https://micropython.org/download/rp2-pico-w/). When I load main.py into pico and disconnect-connect power, pico does not respond to thonny commands (as long as main.py exists). I can only help myself by quickly pressing STOP several times immediately when the pico is connected, only then does the main.py program stop and I can load another program or delete main.py. Older versions supposedly can stop the main.py program at any time (someone on the net says so, I haven't tried it). Can you fix this?

aivarannamaa, 2023-07-02:
Thank you for the report! I suppose it is related to #2836. I'll see what needs to be done.

aivarannamaa, 2023-08-10:
Will be fixed under #2843

closed 2023-08-10 by aivarannamaa

tenapier, 2023-10-13:
Hi, I'm having problems with this issue, losing communication with PicoW once I have a main.py program in it. Any outlook on when this might be fixed. Thank you, Tim