thonny/thonny #2845: esp32 cam Device is busy or does not respond.

PyWebSol, 2023-07-02:
Hello. I am using esp32-cam board (esp 32s). when connected to the phone, the message "Device is busy or does not respond. Your options:

   - wait until it completes current work;
   - use Ctrl+C to interrupt current work;
   - use Stop/Restart to interrupt more and enter REPL."
Tried different firmware. when the board is connected to the arduino ide, all information is displayed, but it is not convenient to use it.

[image]

[image]

aivarannamaa, 2023-07-04:
Thank you for the report!

Is it this board; https://randomnerdtutorials.com/esp32-cam-video-streaming-face-recognition-arduino-ide/ ?

Which Thonny version are you using? The latest in 4.1.1.

You could try different `submit_mode`: https://github.com/thonny/thonny/wiki/MicroPython#advanced-configuration

PyWebSol, 2023-07-04:
the esp32-cam ai thinker board. thonny version 4.1.1. I tried to do what you wrote to me, but it didn't help. on the website https://bipes.net.br/ide / everything works fine

[image]

aivarannamaa, 2023-07-04:
> the esp32-cam ai thinker board

This needs a separate programmer "shield", right? Are you using the one with CH340 chip (https://makeradvisor.com/tools/esp32-cam-mb-usb-programmer/) ?

Beso274, 2023-07-16:
> > the esp32-cam ai thinker board
> 
> This needs a separate programmer "shield", right? Are you using the one with CH340 chip (https://makeradvisor.com/tools/esp32-cam-mb-usb-programmer/) ?

I've got the same problem. I'm using the module in the link you just sent. My error is a little bit longer, here the photo.

[image]

I don't think the problem is in the firmware anymore cause I just updated it to 1.20 ("esp32-20230426-v1.20.0.bin") using the configuration of the interpreter.
My thonny version is the same 4.1.1

aivarannamaa, 2023-07-23:
I think was able to reproduce this problem with another board using the same serial adapter.

I noticed difference in how Pyserial (the library Thonny uses for talking with the device) manages to start a working connection compared to command line tool `tio`. With minimal experiment Pyserial produced a "stuck" connection (no output) the first time after the device was plugged in. After I made a connection with DTR/RTS=False, the device rebooted and this time the connection was working. I could use this approach as a work-around in Thonny.

Unfortunately I did not learn what `tio` does differently so that it doesn't require this reboot to work properly. I did manage to get the device or USB connection into a state where even `tio` could not read anything from the device. Can't explain.

I made these experiments a few weeks ago, so I'm not sure whether it was in Linux or macOS. I believe it was Linux (Ubuntu 23.04).

For Thonny 5.0 I plan to reveal DTR / RTS options in MicroPython configuration page (#2764). Until then you could try whether putting `dtr = False` and `rts = False` directly into ESP32 block of the configuration file helps. See https://github.com/thonny/thonny/wiki/MicroPython#advanced-configuration for instructions.

Beso274, 2023-07-27:
> I think was able to reproduce this problem with another board using the same serial adapter.
>
> I noticed difference in how Pyserial (the library Thonny uses for talking with the device) manages to [...]

Thank you for your response, I tried it, something new has been displayed but i got the same error, here's the screen.

[image]

emperinter, 2023-09-28:
I'm getting the same error, is the configuration change similar to the following? it still fails to connect after I changed config like follows, and the configuration is automatically reset by the software

```ini
[ESP32]
port = /dev/cu.usbmodem544C0050241
interrupt_on_connect = True
sync_time = False
local_rtc = True
restart_interpreter_before_run = True
dtr = False
rts = False
```

[image]

Thonny verion is 4.1.3

Beso274, 2023-09-28:
> I'm getting the same error, is the configuration change similar to the following? it still fails to connect after I changed config like follows, and the configuration is automatically reset by the software
> 
> ```ini
> [ESP32]
> port = /dev/cu.usbmodem544C0050241
> interrupt_on_connect = True
> sync_time = False
> local_rtc = True
> restart_interpreter_before_run = True
> dtr = False
> rts = False
> ```
> 
> [image]
> 
> Thonny verion is 4.1.3 board is esp32-c3

I just pressed the reset button and everything started fine, I haven't changed anything in the configuration file

aivarannamaa, 2023-09-28:
The serial connection to a MicroPython device is sometimes full of wonders for me as well :) You could play with different `dtr` and `rts` values to find out which is best for your platform and MicroPython device.

emperinter, 2023-09-29:
Downgrading Thonny's version and getting the latest beta hardware Fireware is fine with me.