thonny/thonny #2846: internal error after auto-close on file deletion

kr-g, 2023-07-04:
thonny crashes after asking "file deleted. do you want to close the editor too?" and confirmation with "yes". after that opening new files with filebrowser gives the following error.

```
Traceback (most recent call last):
  File "/usr/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/home/benutzer/repo/thonny/thonny/base_file_browser.py", line 595, in on_double_click
    self.open_file(path)
  File "/home/benutzer/repo/thonny/thonny/base_file_browser.py", line 1080, in open_file
    get_workbench().get_editor_notebook().show_file(path)
  File "/home/benutzer/repo/thonny/thonny/editors.py", line 1049, in show_file
    editor = self.get_editor(filename, True)
  File "/home/benutzer/repo/thonny/thonny/editors.py", line 1140, in get_editor
    return self._open_file(filename_or_id)
  File "/home/benutzer/repo/thonny/thonny/editors.py", line 1123, in _open_file
    self.add(editor, text=editor.get_title())
  File "/home/benutzer/repo/thonny/thonny/custom_notebook.py", line 35, in add
    self.insert("end", child, text=text)
  File "/home/benutzer/repo/thonny/thonny/custom_notebook.py", line 46, in insert
    self.select_tab(page.tab)
  File "/home/benutzer/repo/thonny/thonny/custom_notebook.py", line 85, in select_tab
    self.select_by_index(i)
  File "/home/benutzer/repo/thonny/thonny/custom_notebook.py", line 72, in select_by_index
    self.current_page.content.grid_remove()
  File "/usr/lib/python3.10/tkinter/__init__.py", line 2538, in grid_remove
    self.tk.call('grid', 'remove', self._w)
_tkinter.TclError: bad window path name ".!frame.!automaticpanedwindow.!automaticpanedwindow2.!editornotebook.!editor4"

```

kr-g, 2023-07-04:
this applies to thonny running from latest master (rolling baseline)
with 4.1.1 i can not reproduce

kr-g, 2023-07-04:
just merged the lastest master.
now it comes occasionally - but traceback is longer...

```
Traceback (most recent call last):
  File "/usr/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/home/benutzer/repo/thonny/thonny/base_file_browser.py", line 595, in on_double_click
    self.open_file(path)
  File "/home/benutzer/repo/thonny/thonny/base_file_browser.py", line 1080, in open_file
    get_workbench().get_editor_notebook().show_file(path)
  File "/home/benutzer/repo/thonny/thonny/editors.py", line 1049, in show_file
    editor = self.get_editor(filename, True)
  File "/home/benutzer/repo/thonny/thonny/editors.py", line 1140, in get_editor
    return self._open_file(filename_or_id)
  File "/home/benutzer/repo/thonny/thonny/editors.py", line 1123, in _open_file
    self.add(editor, text=editor.get_title())
  File "/home/benutzer/repo/thonny/thonny/custom_notebook.py", line 35, in add
    self.insert("end", child, text=text)
  File "/home/benutzer/repo/thonny/thonny/custom_notebook.py", line 46, in insert
    self.select_tab(page.tab)
  File "/home/benutzer/repo/thonny/thonny/custom_notebook.py", line 85, in select_tab
    self.select_by_index(i)
  File "/home/benutzer/repo/thonny/thonny/custom_notebook.py", line 72, in select_by_index
    self.current_page.content.grid_remove()
  File "/usr/lib/python3.10/tkinter/__init__.py", line 2538, in grid_remove
    self.tk.call('grid', 'remove', self._w)
_tkinter.TclError: bad window path name ".!frame.!automaticpanedwindow.!automaticpanedwindow2.!editornotebook.!editor7"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/home/benutzer/repo/thonny/thonny/base_file_browser.py", line 595, in on_double_click
    self.open_file(path)
  File "/home/benutzer/repo/thonny/thonny/base_file_browser.py", line 1080, in open_file
    get_workbench().get_editor_notebook().show_file(path)
  File "/home/benutzer/repo/thonny/thonny/editors.py", line 1062, in show_file
    self.select(editor)
  File "/home/benutzer/repo/thonny/thonny/custom_notebook.py", line 62, in select
    return self.select_by_index(self.index(tab_id))
  File "/home/benutzer/repo/thonny/thonny/custom_notebook.py", line 72, in select_by_index
    self.current_page.content.grid_remove()
  File "/usr/lib/python3.10/tkinter/__init__.py", line 2538, in grid_remove
    self.tk.call('grid', 'remove', self._w)
_tkinter.TclError: bad window path name ".!frame.!automaticpanedwindow.!automaticpanedwindow2.!editornotebook.!editor7"

```

kr-g, 2023-07-04:
now you can reproduce after right-click on editor panel and "close all"

aivarannamaa, 2023-07-06:
Thanks! I'll reproduce and fix this later.

closed 2023-11-26 by aivarannamaa