thonny/thonny #2848: Not able to install packages in Thonny ide

arbbaza, 2023-07-05:
I am beginner to RP2040 and Circuit python using Thonny IDE.I have tried installing packages using the 'manage packages option' but it does not display any package available.I will share the screenshot below.I tried even installing files manually , but it again displays an error.I am not able to understand what is wrong.
[image]
[image]

aivarannamaa, 2023-07-05:
Usually "an error" is not specific enough to start solving the problem -- it is necessary to include as much detail about the error as possible.

Luckily in the first picture the end of the error message suffices. It looks like you are trying to install the package called `pillow`, but this is not meant for MicroPython.

arbbaza, 2023-07-05:
thank you for the clarification.

closed 2023-07-05 by arbbaza

arbbaza, 2023-07-05:
Another problems i am facing.When i search for any package using the manage packages option,it shows this error.I am unable to find any package or search it.the package i require is adafruit_imageload.Though i have the package downloaded from the circuitPython bundle.I am unable to add it manually ('it says pico write mode disabled' and through manage packages it does not show the avaialble packages at all.
[image]
[image]