thonny/thonny #2849: Thonny IDE not displaying packages at all

arbbaza, 2023-07-05:
I am facing this unique problem,being a beginner to pico and Thonny IDE i am not able to understand what is wrong.When i search for any package using the manage packages option,it shows this error.I am unable to find any package or search it.the package i require is adafruit_imageload.Though i have the package downloaded from the circuitPython bundle.I am unable to add it manually ('it says pico write mode disabled' and through manage packages it does not show the avaialble packages at all.I am attaching the screen shots for reference
[image]
[image]

_Originally posted by @arbbaza in https://github.com/thonny/thonny/issues/2848#issuecomment-1621590301_

aivarannamaa, 2023-07-06:
Please see #1986 for the certificate error.

> unable to add it manually

Do you mean by uploading the files to the board via Thonny's file browser? Can you make the screenshot of this error?

arbbaza, 2023-07-07:
Yes,using the 'Thonny file browser' I am unable to upload the package in the lib folder of Pico and also tried the 'manage packages' option in the IDE,through this method also I was not able to upload packages in Pico.

aivarannamaa, 2023-08-05:
> Yes,using the 'Thonny file browser' I am unable to upload the package in the lib folder of Pico

Do you get an error?