thonny/thonny #2850: how to add ipython as remote ssh interpreter?

paultanger, 2023-07-05:
Hi, I am trying to do something similar to this issue (https://github.com/thonny/thonny/issues/1505), but want to use ipython in a remote SSH connection. I tried to specify the ipython full path, but maybe I am doing it wrong? I am happy to edit the backend if needed - I just had to do that to fix the port number since I'm not using 22.  Thanks.  Here is the error:

```
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/cpython_ssh/cps_back.py", line 213, in <module>
    backend = SshCPythonBackend(**args)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/cpython_ssh/cps_back.py", line 45, in __init__
    self._proc = self._start_main_backend()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/cpython_ssh/cps_back.py", line 153, in _start_main_backend
    return self._create_remote_process(
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/backend.py", line 815, in _create_remote_process
    raise RuntimeError(f"Got {ack!r} instead of expected {PROCESS_ACK!r}")
RuntimeError: Got 'bash: line 1: exec: ipython: not found' instead of expected 'OK'

Process ended with exit code 1.
```

aivarannamaa, 2023-07-06:
Unfortunately you can't get IPython REPL like this. For various reasons Thonny uses its own code for creating the REPL, i.e. what you see in "Shell" pane is an emulation of the regular CPython REPL.

paultanger, 2023-07-06:
ok, thanks!