thonny/thonny #2851: `Upload to /` menu doesn't work

yamamotomas, 2023-07-06:
I have been using Thonny 4.0.2 with Windows 10 (64-bit).
There haven't been any problem in uploading files or directory to Raspberry Pi Pico.

But as soon as I upgraded to Thonny 4.1.1, it cannot upload files to Pico.
The following is the trouble:
* Trying to upload `lib` folder to Pico, I selected the menu of `Upload to /`
* Popup window says "Starting", but it won't finish forever (it can be cancelled).

Does somebody know how to fix this ?
For a moment, I've downgraded the Thonny back to 4.0.2 and workarounded the problem.

[image]
[image]

aivarannamaa, 2023-07-06:
Thank you for the report!

Unfortunately I can't reproduce this in my Windows 11 machine. 

How large is this folder? Can you upload single file? Can you use the Shell without problems? Can you save files via editor without problems?

You could help like this:

1. Put Thonny to debug mode (Tools => Options => General => Debug mode)
2. Restart Thonny
3. Select "Tools => Open Thonny data folder" and check that it contains frontend.log and backend.log
4. Reproduce this error
5. Close Thonny
6. Send me the frontend.log and backend.log

yamamotomas, 2023-07-07:
Thanks for comments.

> How large is this folder? 

The attached image shows the structure of this `lib` folder.

> Can you upload single file? 

I cannot upload even when one file is selected.

> Can you use the Shell without problems? 

The Shell seems to work as usual.

> Can you save files via editor without problems?

Yes, I can save a file to Raspberry Pi Pico.

> Put Thonny to debug mode (Tools => Options => General => Debug mode)

I have attached two log files.

[image]
[frontend.log](https://github.com/thonny/thonny/files/11980817/frontend.log)
[backend.log](https://github.com/thonny/thonny/files/11980818/backend.log)

chechigp, 2023-07-11:
I have the same problem with versions above the 4.02, I'm on Windows 10 64bit too.
After waiting for a while stuck in the uploading window, if I force to cancel and restart Thonny the file seems to be uploaded.

Thanks!

aivarannamaa, 2023-07-22:
Unfortunately I'm not able to reproduce this on Windows 10 either.

Can you please check, whether the first beta of 4.1.0 (https://github.com/thonny/thonny/releases/tag/v4.1.0b1) causes problems for you?

wilare, 2023-07-23:
Hello,

Thank you. The problem is still present with version thonny-4.1.0b1.exe.

Links:
------
[1] https://github.com/thonny/thonny/issues/2851#issuecomment-1646586507
[2] 
https://github.com/

yamamotomas, 2023-07-24:
> Thank you. The problem is still present with version thonny-4.1.0b1.exe.

Same situation to me.

Thinking that anti-virus software (Trend Micro Apex One) might be doing something bad, I've unloaded them just to see the same result.

Thanks.

wilare, 2023-07-25:
I downgraded to 4.02, and the problem was solved. I tried with 4.10b1 and above, and the problem reappeared.
I went back to 4.02, and it works fine. 
I am using Windows 10 Home, version 22H2,  firmware rp2-pico-20230426-v1.20.0.uf2

aivarannamaa, 2023-08-07:
@yamamotomas , @chechigp , @wilare , @iBigDane, I haven't been able to reproduce this problem. I tried reviewing the changes between 4.0.2 and 4.1.0b1 but there are many changes between these and I couldn't spot anything. You could help me by reducing the search scope, though.

[Thonny_snapshots_phase1.zip](https://github.com/thonny/thonny/files/12272343/Thonny_snapshots_phase1.zip) contains three snapshots of Thonny history between versions 4.0.2 and 4.1.0b1. When you select "Tools => Open Thonny program folder", you'll find a directory of `thonny` module. If you close Thonny and replace the content of this directory with one snapshots, then you can check, whether the bug is present in this snapshot or not. It would be really helpful if you could tell me, which of the snapshots introduces the bug (if any).

chechigp, 2023-08-07:
> 

I tried it with no luck. All the snapshots worked when I tried to upload/delete some files to the ESP. For double check I downloaded the last version and the bug reappeared again.
Let me know if I can help with something else.
Thanks for your effort with this project.

aivarannamaa, 2023-08-07:
@chechigp, this is very useful information. The regression either happen after the last snapshot or the problem lies in the updated Python version (3.10.9 vs 3.10.11) or Tk version (8.6.12 vs 8.6.13).

Here is the contents of `thonny` package at [4.1.1](https://github.com/thonny/thonny/files/12274190/thonny_4.1.1.zip). If this works nicely on top of Thonny 4.0.2 installation, then I know I need to investigate Tk or Python changes.

chechigp, 2023-08-07:
> @chechigp, this is very useful information. This indicates the regression either happened after the last snapshot or the problem lies in the updated Python version (3.10.9 vs 3.10.11) or Tk version (8.6.12 vs 8.6.13).
> 
> Here is the contents of `thonny` package at [4.1.1](https://github.com/thonny/thonny/files/12274190/thonny_4.1.1.zip). If this works nicely on top of Thonny 4.0.2 installation, then I know I need to investigate Tk or Python changes.

Well I copied the content of the 4.1.1 package in the 4.0.2 installation folder and the bug appeared.
I had the feeling that maybe I did something wrong and I repeated all the test again but with the following procedure for each test:

1. Disconnect the ESP
2. Uninstall Thonny
3. Fresh Install Thonny 4.0.2
4. Copy the package
5. Start Thonny and connect the ESP
6. Upload a file and delete the file

Well, the behavior was the same. 

- With the 3 _Thonny_snapshots_phase1.zip_ I can upload and delete files to the ESP.
- With the package 4.1.1 copied on the 4.0.2 fresh install the bug appeared again.

Let me know if I help with something more.

aivarannamaa, 2023-08-07:
@chechigp, thanks a lot! This is actually good news -- I'd rather search for problems in Thonny's code than in Python or Tk code.

We now have smaller search space. Please now try next 3 snapshots, this time between 2nd of April and the release of 4.1.0b1: [Thonny_snapshots_phase2.zip](https://github.com/thonny/thonny/files/12279578/Thonny_snapshots_phase2.zip)

chechigp, 2023-08-07:
> @chechigp, thanks a lot! This is actually good news -- I'd rather search for problems in Thonny's code than in Python or Tk code.
>
> We now have smaller search space. Please now try next 3 snapshots, [...]

Ok, maybe we have something. I used the same procedure as before.

- thonny_2023-04-08_59ee0e1830b8c0b2e6549508372ff01a6cbc6114 -> I can't upload anything to the ESP
- thonny_2023-04-15_d704dfe69974eb878735eeb2fac17b1152af6889 -> I can't upload anything to the ESP
- thonny_2023-07-07_41047293d5ec28eb02060704365078b7513a2f20 -> I CAN upload files to the ESP
I tried again the 2023-04-15 commit and I comed back to not be able to upload files to the ESP, the same installation with this commit overwrite with the 2023-07-07 commit and I now I'm able to upload files.

aivarannamaa, 2023-08-07:
@chechigp, thanks, we're getting close! I noticed I had written wrong date for one of the snapshots -- 2023-07-07 should be 2023-04-07. This means 41047293d5ec28eb02060704365078b7513a2f20 (from 2023-04-07) was the last known good commit and 59ee0e1830b8c0b2e6549508372ff01a6cbc6114 (from 2023-04-08) was the first known bad commit. 

Please try one more snapshot between these two: 
[thonny_2023-04-08_16bda02f8f9b3b773973427e9245c6990c262959.zip](https://github.com/thonny/thonny/files/12280570/thonny_2023-04-08_16bda02f8f9b3b773973427e9245c6990c262959.zip)

chechigp, 2023-08-07:
> @chechigp, thanks, we're getting close! I noticed I had written wrong date for one of the snapshots -- 2023-07-07 should be 2023-04-07. This means [4104729](https://github.com/thonny/thonny/commit/41047293d5ec28eb02060704365078b7513a2f20) (from 2023-04-07) was the last known good commit and [59ee0e1](https://github.com/thonny/thonny/commit/59ee0e1830b8c0b2e6549508372ff01a6cbc6114) (from 2023-04-08) was the first known bad commit.
> 
> Please try one more snapshot between these two: [thonny_2023-04-08_16bda02f8f9b3b773973427e9245c6990c262959.zip](https://github.com/thonny/thonny/files/12280570/thonny_2023-04-08_16bda02f8f9b3b773973427e9245c6990c262959.zip)

This commit also works, I'm able to upload files.

aivarannamaa, 2023-08-07:
@chechigp, thanks to you I now have the bad commit (https://github.com/thonny/thonny/commit/59ee0e1830b8c0b2e6549508372ff01a6cbc6114), but I still can't understand what's wrong with it. It tries to avoid the inference from other installations of Thonny, by running Thonny and its back-end processes with Python's `-I` ("isolated") switch and by adding some `sys.path` manipulations.

For a sanity check please see whether you have a folder named `thonny` under `%USERPROFILE%\AppData\Roaming\Python\Python310\site-packages` (e.g. `C:\Users\Aivar\AppData\Roaming\Python\Python310\site-packages\thonny`).

For another test, please take Thonny 4.1.1 and see if things improve when you edit `thonny\running.py` and make the method `can_be_isolated` return `False` instead of `True`.

chechigp, 2023-08-07:
> @chechigp, thanks to you I now have the bad commit ([59ee0e1](https://github.com/thonny/thonny/commit/59ee0e1830b8c0b2e6549508372ff01a6cbc6114)), but I still can't understand what's wrong with it. It tries to avoid the inference from other installations of Thonny, by running Thonny and its back-end processes with Python's `-I` ("isolated") switch and by adding some `sys.path` manipulations.
> 
> For a sanity check please see whether you have a folder named `thonny` under `%USERPROFILE%\AppData\Roaming\Python\Python310\site-packages` (e.g. `C:\Users\Aivar\AppData\Roaming\Python\Python310\site-packages\thonny`).
> 
> For another test, please take Thonny 4.1.1 and see if things improve when you edit `thonny\running.py` and make the method `can_be_isolated` return `False` instead of `True`.

I have 3 python versions but not the 310. No one of these folders have a thonny folder :-S
[image]
I tried to search for the file in my Appdata folder with no luck.
I also tried to install it for "all users" if in case that changes something but no luck.
Any idea about why is missing this thonny folder?

aivarannamaa, 2023-08-08:
> After editing the running.py file returning False in the can_be_isolated method now I CAN upload files to the ESP :-D
So all the problem is related in the python version that thonny use?

If you were running both `True` and `False` variants with the same Python, then it doesn't seem to be related to the Python version. It looks like that something unexpected happens when your Thonny runs the back-end process in isolated mode (with `-I` switch). Unfortunately I can't spot any difference in my machine.

Could you please check (either via Control Panel or command line), whether you have defined any environment variables with `PYTHON`-prefix (e.g. `PYTHONPATH`) in your system? One effect of the `-I` switch is ignoring those. Also, please see whether you have `TCL_LIBRARY` or `TK_LIBRARY` environment variable.

chechigp, 2023-08-08:
> 

I supposed that "isolated mode" meant that use their python interpreter instead of the system python interpreter. That could be another problem because I'm a little lost at how thonny works behind the curtain . ¿^_^

I don't have any environment variable called/including `PYTHONPATH`, `TCL_LIBRARY` or `TK_LIBRARY`. I used the command SET to extract the data and be able to search in an editor.

I list below the `PATH ` entries that mention `PYTHON`.

- C:\Python311\Scripts\
- C:\Python311\
- f:\Programas\Python\Scripts\
- f:\Programas\Python\
- C:\Users\user\AppData\Local\Programs\Python\Python37\Scripts\
- C:\Users\sleep\AppData\Local\Programs\Python\Python37\

aivarannamaa, 2023-08-08:
Here's what "isolated" (`-I`) means: https://docs.python.org/3/using/cmdline.html#cmdoption-I

After re-reading this, it occurred to me that I had forgotten that Thonny itself sets some `PYTHON*` environment variables for the backend process, which now get ignored because of `-I`. If you still bear with me, then please take Thonny with  `can_be_isolated` returning `False` (i.e. the way it worked for you) and (in addition to previous change) comment out following line: 
https://github.com/thonny/thonny/blob/8a8e35c3b38661b3885d40a58e68aa6a4496efff/thonny/running.py#L1033

If this makes the upload fail again, then I know I need to investigate the encoding issues. It could be helpful if you sent me the file you were trying to upload. I did try uploading a file with non-ascii chars in it, but maybe this test needs to be more specific.

aivarannamaa, 2023-08-10:
I was now able to reproduce the problem as indicated by the logs from @wilare -- in this case the problem was related to non-ascii characters in the source file paths. I could solve it by getting rid of the `-I` switch and thus restoring the effect of `PYTHONIOENCODING` environment variable.

I still can't explain the logs submitted by @yamamotomas -- there are no error messages, but only 10 of the submitted 14 files were uploaded and the process seemed to block after this.

Anyway, I'll release the fix soon, so that you all can more easily check if the problem gets solved in your case.

aivarannamaa, 2023-08-11:
Please test https://github.com/thonny/thonny/releases/tag/v4.1.2

closed 2023-08-11 by aivarannamaa