thonny/thonny #2854: Thonny loses control while testing ESP32 watchdog

TomLinz, 2023-07-07:
With a simple piece of code I wanted to test the inbuilt watchdog feature of the ESP32, using  Thonny as usual:
1   #ESP32 Watchdog tester
2   from machine import WDT
3   import time
4   
5   wdt = WDT(timeout = 10000) #enable watchdog with 10 second timeout
6   wdt.feed()
7   
8   print ('Restart Watchdog test')
9   
10   i = 0
11   while 1:
12        print (i)
13        i = i + 1
14        time.sleep(1) #sleep 1 second

 The watchdog worked fine, repeatedly. But getting rid of this code running from main.py on board the ESP32 was not easy. I tried the usual <ctrl-d> and <ctrl-F2>, but seldom did anything happen. And if I managed to get the REPL prompt for a while, the ESP32 started again bombing with its never ending output stream, making it impossible to upload something else.
 I thought I should press the reset button of the ESP32 board, and release it at a correct moment to upload a friendlier code into main.py. However, the reset button is not easy to reach in my specific hardware.
 After a long tweaking I finally managed to replace the code in main.py, with another peaceful Python code.  This required a careful timing and prompt action.
 Just wondering if there is an effective method to stop the MicroPython device in all circumstances?

TomLinz, 2023-07-07:
Okay, this system does not like some characters. I wanted to say: '...tried the usual ctrl-D and cntrl-F2...'.

aivarannamaa, 2023-08-05:
Sorry for the late reply!

As far as I know, it is common struggle with different boards. Some set up a delay (`time.sleep(...)`) as the first step in the program, which provides a window of time for interrupting it.

TomLinz, 2023-08-07:
Hello Aivar

Thanks for your answer. I see, yes, best to insert more delay.

I am quite happy usíng Thonny daily. But to optimize using it, can you point me
to a user manual or something similar?

Best regards,
Tom

-----------------------
Tom Lindeman
City Network Finland Oy
Cygnaeuksenkatu 7 D 36
00100 Helsinki
Finland
Tel: +358 400 413 244
E-mail: ***@***.***

aivarannamaa, 2023-09-27:
Sorry for the late reply!

> But to optimize using it, can you point me to a user manual or something similar?

There's something if you select "Help contents" from the "Help" menu.

closed 2023-09-27 by aivarannamaa