thonny/thonny #2856: Thonny 4.1.1 Chinese version has a problem

ZJUPrily, 2023-07-08:
In Thonny 4.1.1, if I set the language to Simplified Chinese, many options in the menu cannot be translated into Chinese, and only some options can be translated into Chinese. In addition, on Win11, the menu icons and buttons become very small and cannot be seen clearly.

aivarannamaa, 2023-08-05:
Sorry for the late reply!

The translations are made by volunteers. If something is missing then you can enhance it yourself at https://poeditor.com/join/project/Gh188fdYH6

> In addition, on Win11, the menu icons and buttons become very small and cannot be seen clearly.

What is the resolution of your monitor?

ZJUPrily, 2023-08-05:
> Sorry for the late reply!
>
> The translations are made by volunteers. If something is missing then you can enhance it yourself at https://poeditor.com/join/project/Gh188fdYH6
>
>
> What is the resolution [...]

The resolution of my monitor is 2500 * 1600. I later solved this problem by adjusting the UI scaling factor to 2.5. For the issue of incomplete Chinese translation, it is suggested to replace the Chinese package of Thonny 4.1.1 with the Chinese package of Thonny 4.1.0b1 and previous versions.

closed 2023-08-10 by aivarannamaa