thonny/thonny #2858: Trying to install Thonny non-bundle into Python 3.11.4: Failed building wheel for thonnyapp

nestorquay, 2023-07-09:
I have a fresh copy of Python 3.11.4 installed on my Windows 10 machine.  My PATH includes C:\Program Files\Python-3-11-4\Scripts\. I'm running "pip install thonnyapp" from an Administrator command line.  When installing Thonny 4.1.1 alone without the Python bundle into my existing Python 3.11.4, I get the following errors:

1)  Building wheel for thonnyapp (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for thonnyapp (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [75 lines of output]

and

2)        File "C:\Users\Mike\AppData\Local\Temp\pip-build-env-kjx0t7vh\overlay\Lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "<string>", line 69, in run
        File "<string>", line 43, in post_install
        File "<string>", line 61, in windows_post_install
        File "<string>", line 58, in create_windows_shortcut
        File "<string>", line 16, in get_thonny_resource
      ModuleNotFoundError: No module named 'thonny'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for thonnyapp
Failed to build thonnyapp

Please advise.  Thanks

aivarannamaa, 2023-08-11:
This is caused by the changes in newer `pip` versions.

I think I should retire `thonnyapp`, because it is getting more difficult to create packages, which have some extra steps in the end of the installation process. 

As an alternative, I recommend `pip install thonny` and manually creating the shortcut to `thonny` executable in the `Scripts` directory.

Also, just in case you didn't know -- you don't need to install Thonny and Python separately in order to use another Python interpreter for running your programs. You could let the bundled interpreter run only Thonny's GUI and select another interpreter for running the programs.

aivarannamaa, 2023-08-11:
I now yanked the package and updated Wiki accordingly: https://github.com/thonny/thonny/wiki/Windows#installing-thonny-and-python-separately

closed 2023-08-11 by aivarannamaa