thonny/thonny #2860: Freshly installed App crashes on save attempt during first run if configuration.ini is missing

mihelm, 2023-07-11:
Hi

I checked the open issues, but didn't find anything that struck me as a duplicate — my apologies if I didn't look hard enough.

I've come across a reproducible crash of version 4.1.1 — I have not tested other versions.

I've only tested it on the current version of macOS, running on Apple Silicon.

The crash manifests like this: When one launches a freshly installed app, and tries to save the open file (content doesn't matter), the app crashes before even showing the file picker. From the second launch onwards, it works without issue.

I was able to track it down to the configuration.ini file in `~/Library/Thonny — if this file exists (even if the original has been deleted and replaced with an empty file using touch), the crash doesn't occur. If the app has run before, but that file is deleted, the crash will occur on the next launch.

If the file is missing, the app creates it after launch. But it still crashes.

Cheers, michael.

aivarannamaa, 2023-07-23:
Thank you for the report and sorry for the late reply!

That's really useful information. I was able to reproduce this.

I have experienced such crash inside Tk (Thonny's UI toolkit) with new installations, but I didn't have any specific theories.

I'll investigate the difference in Thonny's internal state during first and subsequent runs and hopefylly I can find a work-around.

aivarannamaa, 2023-08-05:
Weird... I can't reproduce this anymore with the same installation of 4.1.1

aivarannamaa, 2023-08-11:
Nevermind, I had tried to reproduce it incompletely. I now discovered that instead of saving an editor, opening a menu also suffices to induce the crash.

aivarannamaa, 2023-08-12:
I found out that it is a known bug in TkAqua: https://sourceforge.net/p/tcl/mailman/message/37882923/ and https://core.tcl-lang.org/tk/info/ef5d3e29a4

I'll refactor the start-up to avoid this.

aivarannamaa, 2023-08-12:
The fix will be in Thonny 4.1.3.

closed 2023-08-12 by aivarannamaa