thonny/thonny #2861: Thonny refuses to import network in Micropython for Pico

amh209, 2023-07-12:
This is still a very important problem. The Pico W cannot connect to the Wifi because Thonny returns a "Traceback (most recent call last):
  File "<stdin>", line 9, in <module>
ImportError: no module named 'network'" 
error
[image]

aivarannamaa, 2023-07-13:
It's MicroPython, which returns this error, not Thonny. Are you sure you installed "Pico W" MicroPython, not the one meant for Pico?

amh209, 2023-07-13:
The Pico W firmware is v1.20.0.  The interpreter is Pico - how would I install a Pico w interpreter?

aivarannamaa, 2023-07-13:
When I connect my Pico W, then I see following information:

```
MicroPython v1.20.0 on 2023-04-26; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>> 
```

What do you get?

amh209, 2023-07-13:
with no pico connected - 
MicroPython v1.17 on 2021-09-02; Raspberry Pi Pico with RP2040

Type "help()" for more information.
 Plugging in a pico board (not W) give the same as above and the USB connection audio ping.
(MicroPython v1.17 on 2021-09-02; Raspberry Pi Pico with RP2040
Type "help()" for more information.)

The Pico W doesn't initiate a USB connection unless the Bootsel button is pressed - am now thinking there is a problem with the board...

The INFO_UF2.TXT is 
UF2 Bootloader v3.0
Model: Raspberry Pi RP2
Board-ID: RPI-RP2
and I've loaded the v1.20.0 uf2 more than once so now stuck...

aivarannamaa, 2023-07-13:
> MicroPython v1.17 on 2021-09-02; Raspberry Pi Pico with RP2040

This is meant for "plain" Pico, not for Pico W. Please note that in my case it says "Raspberry Pi Pico **_W_** with RP2040"

RajSahani23, 2023-10-01:
I found the same error while installing the network module in Thonny IDE for Raspberry Pi Pico, because I'm using external wifi module. Please help me to fix this problem!!
[image]

stereoIII6, 2023-11-04:
same issue

aivarannamaa, 2023-11-16:
@RajSahani23, the proper Pico W variant of MicroPython already contains `network` module. You can't fix installing wrong variant of MicroPython by installing package named "network" (which is meant for CPython)

PeimanD, 2024-01-05:
I had the same problem and realized I had loaded the wrong UF2 file into my pico w so then Thonny was not recognizing it as pico W but a Pico hence installing the wrong micro Python on the unit missing the network package. 

Get the UF2 file from here: https://www.raspberrypi.com/documentation/microcontrollers/micropython.html. mode the file into the unit on boot mode nad wait for it to restart again.