thonny/thonny #2862: Unable to save micropython code to psoc6 board using Thonny Idea

NikhitaR-IFX, 2023-07-13:
Hi Team,

We have issue while trying to save a MicroPython script on filesystem (FS) to Psoc6 board. The file size is 16384 bytes. Trying to save the same file to same board using Arduino Lab for MPY as well as mpremote tool works perfectly fine, which ensures is not the problem with available memory on FS.

Additional information: I can save multiple exact 16 KB files to FS using thonny perfectly fine. 

Attached is the error message.
[image]

Thanks and Regards,
Nikhita

aivarannamaa, 2023-07-23:
Thank you for the report! Which MicroPython (or CircuitPython) version are you using? From where did you download it? Which Thonny version are you using?

NikhitaR-IFX, 2023-07-23:
Hi @aivarannamaa ,

I am using the latest version of micropython and this is continuously synced with the changes in the main repository. We have developed a new port for psoc6 (https://github.com/Infineon/micropython) and have been using thonny for saving the script to filesystem so far.  

I am currently at version 4.1.1 for thonny. But I did also test this in the last version I had - 4.1.0 and it was failing their as well.  

Please let me know if you need any more details and I can provide you the same. Thank you for your support!

Best Regards,
Nikhita

aivarannamaa, 2023-08-05:
How much space does your device have? Does this problem occur also when you first delete old file and then upload the new version?

NikhitaR-IFX, 2023-08-08:
Hi @aivarannamaa ,

The device has 512MBit external flash but I don't suspect this to be an issue with space since I could upload the same file using mpremote perfectly fine and yes the problem does exist even if the FS was initially free and loaded freshly with this file using thonny. Furthermore, as raised by one of our users, the problem continues still. 

It would be helpful to know, how is the IDE handling the integration of MicroPython and boards briefly to understand if we missed out some steps there. Why is this working differently on different IDEs?

Thank you for your support.

Thanks and Regards,
Nikhita

aivarannamaa, 2023-09-27:
Sorry for the late reply!

Thonny saves files by sending commands like `_temp_file = open(filename, "w")`, `_temp_file.write(b".....")`, `_temp_file.close()` to MicroPython REPL. I believe it's not much different compared to e.g. mpremote.

The common problem is that serial connection is not always reliable channel. There can be subtle bugs in the OS level, in USB driver, in device's firmware and also on MicroPython. Quite often the bugs become apparent with only very specific timing, so it's hard to reproduce them. That's my main problem with developing MicroPython support for Thonny.

Of course, there are also bugs in Thonny, but it's hard to tell them apart if I'm not able to reproduce the problem. I'll consider bying the board like yours which may help me find faults in Thonny.

Can you give me some pointers for buying a board for reproducing this issue?

NikhitaR-IFX, 2023-09-28:
Hi @aivarannamaa,

Thank you for looking into this issue. Yes you are right that mpremote as well sends commands serially. 
In the event you want to get hold of this board, you can checkout different vendors from here: [product link](https://www.infineon.com/cms/en/product/evaluation-boards/cy8cproto-062-4343w/) and you can find the micropython port of ours [link](https://github.com/Infineon/micropython/tree/ports-psoc6-main/ports/psoc6) here.

If you need anymore details, let me know and I will be happy to share.

Again, thanks for your help!

Thanks and Regards,
Nikhita