thonny/thonny #2863: Searching for a package causes an error

ntate6630, 2023-07-13:
Running Thonny 4.1.1
Python 3.8.10
Tk 8.6.9
Windows 7 - 32bit.

Traceback (most recent call last):
  File "C:\Users\nicktate\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1892, in __call__
    return self.func(*args)
  File "C:\Users\nicktate\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 814, in callit
    func(*args)
  File "C:\Users\nicktate\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\pip_gui.py", line 698, in poll_fetch_complete
    results = results_future.result()
  File "C:\Users\nicktate\AppData\Local\Programs\Thonny\lib\concurrent\futures\_base.py", line 437, in result
    return self.__get_result()
  File "C:\Users\nicktate\AppData\Local\Programs\Thonny\lib\concurrent\futures\_base.py", line 389, in __get_result
    raise self._exception
  File "C:\Users\nicktate\AppData\Local\Programs\Thonny\lib\concurrent\futures\thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "C:\Users\nicktate\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\pip_gui.py", line 155, in _fetch_search_results
    mp_org_result = self._perform_micropython_org_search(query)
  File "C:\Users\nicktate\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\pip_gui.py", line 181, in _perform_micropython_org_search
    data = self._get_mp_org_index_data()
  File "C:\Users\nicktate\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\pip_gui.py", line 97, in _get_mp_org_index_data
    self._mp_org_index_data = json.load(fp)
  File "C:\Users\nicktate\AppData\Local\Programs\Thonny\lib\json\__init__.py", line 293, in load
    return loads(fp.read(),
  File "C:\Users\nicktate\AppData\Local\Programs\Thonny\lib\json\__init__.py", line 343, in loads
    s = s.decode(detect_encoding(s), 'surrogatepass')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x8b in position 1: invalid start byte

aivarannamaa, 2023-07-22:
Thank you for the report!

Does it happen for a specific search term or for any search term? Can you give an example?

aivarannamaa, 2025-05-27:
Search has been refactored by now so this case is obsolete.

closed 2025-05-27 by aivarannamaa