thonny/thonny #2865: The postion of the command line ...

boa2145, 2023-07-14:
Hello, Is it possible to change the position of the command line inside Thonny's IDE? I don't like the command line at the bottom, but I prefer to use the comman line on right side that I can code on the left. Would appreciate feedback! Regards ... Gregor

SimUKdev, 2023-07-14:
You can adjust the layout following the [Custom Layout](https://github.com/thonny/thonny/wiki/Custom-layout) guide in the Wiki.

If you use `shellview.location = e` that will make the Shell (command line) appear on the right hand side.

Some others that I use to customise my Thonny layout are:

```
variablesview.location = ne
outlineview.location = ne
assistantview.location = e
helpview.location = se
todoview.location = se
stackview.location = s
filesview.location = s
notesview.location = s
eventsview.location = s
astview.location = s
objectinspector.location = s
```

boa2145, 2023-07-14:
Thanks for your reply, SimUKdev. I changed Thonny's configuration file and it works now. I see the command line on the right. That's much better than at the bottom. Regards ... Gregor

closed 2023-07-15 by aivarannamaa