thonny/thonny #2868: Run with F5?

AakeJon49, 2023-07-18:
I can't run my scripts with F5?

aivarannamaa, 2023-07-19:
> I can't run my scripts with F5?

What does it mean? Is the command disabled? Or do you get an error? Or nothing happens? 

Are you using CPython or MicroPython back-end?

Which Thonny version are you using?

AakeJon49, 2023-07-19:
If I press F5 on my computer, nothing happens.
If I click the Run-button or go to Run menu and click run, it works
perfectly.

[image: image.png]

Den ons 19 juli 2023 kl 09:23 skrev Aivar Annamaa ***@***.***

>:

> I can't run my scripts with F5?
>
> What does it mean? Is the command disabled? Or do you get an error? Or
> nothing happens?
>
> Are you using CPython or MicroPython back-end?
>
> Which Thonny version are you using?

Moosems, 2023-07-19:
He needs more info than that to fix the issue. Can you show video? What's the system? What version? There's a lot of things at play in Thonny so it's important you give us more info so we can help you.

Moosems, 2023-07-19:
What Python version and distro do you use?

Moosems, 2023-07-19:
Does F5 work elsewhere or is it your computer?

aivarannamaa, 2023-07-20:
Do you have Fn key on your keyboard? Try pressing Fn+F5.

AakeJon49, 2023-07-20:
I'm sorry to answer so late but I have been on a journey.
I made a new test now and everything seems to be working.
I can't say what was wrong before but maybe the Ubuntu update that I
installed this morning and just made a restart on the computer, made the
thing.
Sorry for taking your time.
Thank you for your answering and for a very good IDE.

Den tors 20 juli 2023 kl 06:30 skrev Aivar Annamaa ***@***.***

>:

> Do you have Fn key on your keyboard? Try pressing Fn+F5.

closed 2023-07-21 by aivarannamaa

AakeJon49, 2023-07-21:
Great!

Den fre 21 juli 2023 kl 14:23 skrev Aivar Annamaa ***@***.***

>:

> Closed #2868 <https://github.com/thonny/thonny/issues/2868> as completed.
>
> <https://github.com/thonny/thonny/issues/2868#event-9884783595>, or
> unsubscribe