thonny/thonny #2869: Command output was not empty

Olfuksss, 2023-07-18:
Succesfully initialized picozero-0.4.2
Starting to apply changes to the target.
Copying picozeroError  (ManagmentError ( . . . ), 'Command output was not empty')

i have this error while trying to download picozero package

aivarannamaa, 2023-07-22:
Thank you for the report!

Which OS are you using? Which Thonny version? Which MicroPython version?

Olfuksss, 2023-07-23:
Windows 10, thonny 4.1.1, python 3.10.11

aivarannamaa, 2023-11-13:
Sorry for the late reply!

If you are still having this problem, then please try Thonny 4.1.3. If this doesn't fix it, then please send me the content of the backend.log (Tools => Open Thonny data folder).

aivarannamaa, 2023-12-26:
@rodolfobsantos , thank you for the log and sorry for the late reply!

The log mentions OSError 28, which can mean being out of flash space or having only fragmented space on the flash (https://github.com/micropython/micropython/issues/3121).

Regarding openai and pydantic-core -- are you sure these libraries are meant for MicroPython?

rodolfobsantos, 2023-12-29:
Hi @aivarannamaa , I appreciate your support!

AFAIK these libraries are not meant for MicroPython, but I was trying to deploy them to be able to use some specific functions that are contained in the openai library to call for their APIs directly.

aivarannamaa, 2024-02-24:
I assume your device simply ran out of flash space.

closed 2024-02-24 by aivarannamaa

Don-Kent, 2024-04-02:
Hi @aivarannamaa , I have the same issue "Command output was not empty". I'm running Thonny 4.1.4 on Windows 11 with Python 3.10.11 whilst trying to add the drv8830 Motor Driver package for use with a RPi Pico, I've attached the backend.log.
Also I'm getting the same message for any package I try to add!
[backend.log](https://github.com/thonny/thonny/files/14834506/backend.log)