thonny/thonny #2870: Plotter scroll speed

cuedcad, 2023-07-21:
I am sorry if this is the wrong place to ask the question, but I wondered
if there is a option to change the scroll speed of the plotting window and/or number of
plots to be plotted in the window. Ideally
dynamically, e.g. up/down hotkeys

aivarannamaa, 2023-07-21:
Plotter speeds depends on your program's output speed. If you want to make plotting slower, then you could use  `time.sleep(...)` between your print-s.

cuedcad, 2023-07-21:
Thank you for the fast reply
Yes sleep is a option, but awkward/time consuming to set up if you 
are viewing/just checking multiple data streams which have different 
or varying 'periodicities'**.  If there was an option to increase/decrease the
 time scale/number of pixels between updates it would be very helpful.

  NB ** This approach is also useful for investigating dtat where the printing 
rate/frequency is non constant (eg perhaps dependent triggers)

Many Thanks

aivarannamaa, 2023-07-21:
So you want to control the horizontal pixel distance between the data points in the plotter?

cuedcad, 2023-07-23:
<..control` the horizontal pixel distance between the data points in the plotter...>

 Yes ideally. Currently even expanding the plotter window/pane in x seems to
only plot the same number of points rather than expand the viewed data

  [If someone is modding the code, a possible addtionaly useful feature
would be able to display a vertical/x grid (e.g.  like the current dashed y-scale grid)
  - either every n data points. 
  - or perhaps if including an optional value in the print statement (allowing the labelling
 of x axis), e.g. print ([xaxis=1.0s], xaccel, yaccel)  resulting in a xlabel of '1.0s'.

[image]

Best Wishes