thonny/thonny #2871: Mouse cursor isn't turning into text cursor in the IDE

boraturanli, 2023-07-23:
My mouse pointer isn't turning into the "vertical line" that it normally should in the terminal. It makes text selection cumbersome. How can I fix this?

macOS

aivarannamaa, 2023-08-05:
Thank you for the report! Which Thonny version are you using? Where does the wrong cursor show? In the Shell or editor?

boraturanli, 2023-08-06:
Thanks for the reply!

I'm using Thonny 4.1.1 on macOS 13.4.1 with Python 3.10.11

This happens only in the editor, the Shell works fine.

shawnphelps, 2023-10-21:
This bug is still present with Thonny 4.1.3 on Windows 10.

After opening Thonny, the first time the cursor enters the editor, it will correctly switch to a text selection cursor. However, once the cursor exits the editor region, it will no longer switch to the text selection cursor when re-entering the editor region.

Clicking on the new button (not pressing Ctrl+N) or opening an existing file will reset the editor region so that the cursor will correctly switch the first time it enters the editor region. Pressing Ctrl+N with the cursor over the editor region will briefly flash the correct (text selection) cursor and then return to the default cursor.

farhan1666, 2024-09-18:
This error is still present on Thonny 4.1.6 exactly as described above on Windows 11