thonny/thonny #2872: Thonny: certificate verify failed: certificate has expired (_ssl.c:1007)>

EdwardPalaci, 2023-07-24:
Error Trying to install ssd1306, from tools/manage packages:
Could not fetch search results:
<urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1007)>

petra20, 2023-07-25:
I am also getting this error when trying to install picozero

urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1007)>

aivarannamaa, 2023-07-25:
Please see https://github.com/thonny/thonny/issues/1986

petra20, 2023-07-25:
So I have read #1986 and downloaded lets-encrypt-r3.der, which I agree does not sound like the best way to fix this problem, but needs must.
But, what do I need to do now I have it, I have tried google, but !!!!!

petra20, 2023-07-25:
Woohoo!!!
Followed: 
https://support.securly.com/hc/en-us/articles/360026808753-How-do-I-manually-install-the-Securly-SSL-certificate-on-Windows
And now I can flash the LED.
Happy dayz 😎

closed 2023-07-26 by aivarannamaa

EdwardPalaci, 2023-07-26:
Glad, it worked for you,I have windows 11 and the menu sequence is
different. Thanks!