thonny/thonny #2875: Thanks for understanding.

3469134108, 2023-07-27:
I don't think authors should bring their own political views and positions to open source projects. (personal views only)

nesergen, 2023-07-28:
Agree. Politics is a dirty and dark matter, incomprehensible to ordinary people. We do not know much to draw absolutely correct conclusions.
If the author is firm in his position, then I ask you to make it possible to independently (in the settings) remove the flag of Ukraine from the interface, it is annoying.

sosi-deadeye, 2023-08-03:
I always remove it manually. Btw. support projects, which do not show any flags.

Moosems, 2023-08-03:
Honestly, regardless of my opinions on any conflict, I don't really feel a need for an open source project to ayate given opinions. That being said: if the owner (who I admire) wants to keep it, I will support him regardless. If you don't like seeing a flag, that's a small price to pay for such an amazing project.

Pernat1y, 2023-08-14:
> Politics is a dirty and dark matter, incomprehensible to ordinary people. We do not know much to draw absolutely correct conclusions.
Here goes another "we are out of politics" and "we don't know all the truth" BS. That's, like, russia propaganda's bingo.

nesergen, 2023-08-14:
> Here goes another "we are out of politics" and "we don't know all the truth" BS. That's, like, russia propaganda's bingo.

Do you really believe that only Russians know how to lie? Those. there is no propaganda in America, there is no propaganda in Europe either, and in Ukraine all the saints also speak exclusively the truth? And all the media are incorruptible and strictly observe professional etiquette? Yes, you live in a perfect world!

OlehAhieienko, 2023-08-14:
> Do you really believe that only Russians know how to lie? Those. there is no propaganda in America, there is no propaganda in Europe either, and in Ukraine all the saints also speak exclusively the [...]

Again, [Whataboutism](https://en.wikipedia.org/wiki/Whataboutism)

aivarannamaa, 2023-08-14:
> there is no propaganda in Europe either, and in Ukraine all the saints also speak exclusively the truth? And all the media are incorruptible and strictly observe professional etiquette?

Are you implying that the main reason people are trying for support Ukraine, is the propaganda of "the collective west"? Not the invasion itself? Are you trying to convince me that Ukraine is not worth supporting in this situation? If not then what? Should I reword my call and replace "invasion" with "special military operation" to make it less hurtful for Russians?

3469134108, 2023-08-16:
> > there is no propaganda in Europe either, and in Ukraine all the saints also speak exclusively the truth? And all the media are incorruptible and strictly observe professional etiquette?
> 
> Are you implying that the main reason people are trying to support Ukraine, is the propaganda of "the collective west"? Not the invasion itself? Are you trying to convince me that Ukraine is not worth the support, because without western propaganda we would see that the invasion was justified?
> 
> If not, then what? Should I reword my call and replace "invasion" with "special military operation" to make it less hurtful for Russians?

No, I don't have those intentions. I'm a 16 pair student and I don't have any political views.

3469134108, 2023-08-16:
Declare again! I'm just a junior high school student. Do you really think I'm speaking up for Ukraine or Russia? No, you're wrong. This is just a suggestion. There's no political guidance content.

closed 2023-10-25 by aivarannamaa

nesergen, 2026-01-17:
@gnull Thanks, that is grate !