thonny/thonny #2877: Command output was not empty

Olfuksss, 2023-07-28:
Thonny Version: 4.1.1
OS: Windows 10
Python: Python 3.10.11
Tk: Tk 8.6.13
Error code below: 
`install picozero
Looking in indexes: http://127.0.0.1:36628
127.0.0.1 - - [28/Jul/2023 22:56:00] "GET /picozero/ HTTP/1.1" 200 -
Collecting picozero
  Using cached picozero-0.4.2-py3-none-any.whl
Installing collected packages: picozero
Successfully installed picozero-0.4.2
Starting to apply changes to the target.
Copying picozeroError (ManagementError(...), 'Command output was not empty')
`

aivarannamaa, 2023-08-11:
Assuming it's the same problem as #2869

closed 2023-08-11 by aivarannamaa

Olfuksss, 2023-08-12:
> Assuming it's the same problem as #2869

Yeah and you didnt helped🤡

clear-code-projects, 2024-05-28:
I am having the same issue. I am on windows 11 and Thonny 4.1.4; has there been a fix yet? 

error code: 
install --progress-bar off picozero
Looking in indexes: http://127.0.0.1:36628
127.0.0.1 - - [28/May/2024 19:58:20] "GET /picozero/ HTTP/1.1" 200 -
Collecting picozero
  Using cached picozero-0.4.2-py3-none-any.whl
Installing collected packages: picozero
Successfully installed picozero-0.4.2
Starting to apply changes to the target.
Copying picozeroError (ManagementError(...), 'Command output was not empty')