thonny/thonny #2880: Wiki: installing Thonny and Python separately for Windows

rootsmusic, 2023-07-30:
I suggest starting the [instructions for Windows](https://github.com/thonny/thonny/wiki/Windows#installing-thonny-and-python-separately) with a Python download from the Microsoft Store, then creating a virtual environment for Thonny under that Python version.

aivarannamaa, 2023-08-11:
Because of updates in `pip` (and other trends in Python packaging) it's getting harder to support packages with extra steps in the end of the installation process. Therefore I now yanked `thonnyapp` and updated Wiki accordingly https://github.com/thonny/thonny/wiki/Windows#installing-thonny-and-python-separately.

closed 2023-08-11 by aivarannamaa

rootsmusic, 2023-08-11:
Ok @aivarannamaa, also see #2874 for non-Windows platforms.