thonny/thonny #2881: SSL expired - can't search for any packages

budley, 2023-07-30:
Already a closed issue, but seems it has come back? Hope it's OK to post again about it.
Newcomer to Thonny here.
Tools | Manage Packages
Search (for anything)
"Could not fetch search results: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1007)"

jeamigue, 2023-07-31:
It is likely an error outside of Thonny. Could you elaborate on what OS you are using? Certificates errors can be caused by many different factors. Maybe check first that you did not modify the time of the OS manually.

It is probable Thonny cannot do anything regarding this, besides disabling certificate validation [which is not a good option](https://access.redhat.com/articles/2039753).

budley, 2023-07-31:
Hi,

Thanks for coming back to me on this.

I’m using Windows 10 (latest updates)
Not using a VPN or anything like that.
Usual Windows firewall.
OS is running sync’d to “real time” in UK.
No fancy anti-virus (just defender)

I don’t know which URLs you reference for the libraries but
https://pypi.org/
https://docs.micropython.org/
all have OK certificates in a browser
[A screenshot of a computer  Description automatically generated] [A screenshot of a computer  Description automatically generated]
My default browser is Brave.

I added “hole” through firewall explicitly for Thonny:
[A blue and white rectangular object with orange lines  Description automatically generated]

I tried running as administrator.

Still the same problem.

I’ll leave that with you, if I can give you any more info just shout.

Cheers,
Gerald