thonny/thonny #2884: SD Card Access via Circuitpython on Adafruit Feather M0 Adalogger

MariusTOR, 2023-08-01:
I am attempting to run the following code as a prelude to writing data to an SD card on an Adafruit Feather M0 Adalogger.  This is part of a larger code, but, if I run just this snippet, I get the error below.  If I run the whole code, the device disconnects.

I recently installed Circuitpython 8.2.2 on the Feather, and am using Thonny to access it.

`import time`
`import board`
`import busio`
`import adafruit_sdcard`
`import storage`
`from digitalio import DigitalInOut`

`# initialize the I2c bus`
`i2c = busio.I2C(board.SCL, board.SDA)`

`# initialize SPI bus and set CS pin`
`spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)`
`# sd card reader CS pin on Feather M0 Adalogger is connected to board.D4`
`cs = DigitalInOut(board.D4)`

`# initialize sdcard and storage`
`sdcard = adafruit_sdcard.SDCard(spi, cs)`
`vfs = storage.VfsFat(sdcard)`
`# mount the storage device`
`storage.mount(vfs, "/sd")`

Upon running this code snippet, I get the following error message:

`ERROR   thonny.plugins.micropython.backend: PROBLEM WITH THONNY'S BACK-END:
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 347, in _handle_normal_command
    response = handler(cmd)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 695, in _cmd_execute_source
    self._execute(source, capture_output=False)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 565, in _execute
    self._execute_with_consumer(script, self._send_output)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 700, in _execute_with_consumer
    self._forward_output_until_active_prompt(output_consumer)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 782, in _forward_output_until_active_prompt
    self._check_for_side_commands()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 660, in _check_for_side_commands
    self._submit_input(cmd.data)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 519, in _submit_input
    self._write(block)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 533, in _write
    self._connection.write(data)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\serial_connection.py", line 85, in write
    assert size == len(data)
AssertionError
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 347, in _handle_normal_command
    response = handler(cmd)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 695, in _cmd_execute_source
    self._execute(source, capture_output=False)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 565, in _execute
    self._execute_with_consumer(script, self._send_output)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 700, in _execute_with_consumer
    self._forward_output_until_active_prompt(output_consumer)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 782, in _forward_output_until_active_prompt
    self._check_for_side_commands()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 660, in _check_for_side_commands
    self._submit_input(cmd.data)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 519, in _submit_input
    self._write(block)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 533, in _write
    self._connection.write(data)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\serial_connection.py", line 85, in write
    assert size == len(data)
AssertionError`

aivarannamaa, 2023-09-23:
Thank you for the report and sorry for the delay!

Which Thonny version are you using?

The stacktrace gives the impression that you have entered something in the shell after running the script. Is this correct?

MariusTOR, 2023-09-24:
Peace!

No problem -- thanks for writing back. For better or worse, I gave up and used the Arduino IDE instead. However, to answer your question, I just tried it again...and it worked. I did update to the newest version of Thonny. I believe that I had v4.0.0 running when I tried it before and this failed. I now have version 4.1.2 of Thonny and reinstalled CircuitPython (v8.2.6 now) on the Adalogger, and the code below produced a text file on the SD card with the intended text. I guess it got fixed in the version roll?

```
import adafruit_sdcard
import busio
import digitalio
import board
import storage

# Connect to the card and mount the filesystem.
spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
cs = digitalio.DigitalInOut(board.SD_CS)
sdcard = adafruit_sdcard.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")

# Use the filesystem as normal.
with open("/sd/test.txt", "w") as f:
    f.write("Hello world\n")
```

aivarannamaa, 2023-09-25:
Thanks! 

I remember fixing similar issue, so we can call it done.

closed 2023-09-25 by aivarannamaa