thonny/thonny #2885: Cannot find MicroPython PyPI packages

JackAtKitronik, 2023-08-02:
Hi Thonny team,

Since a recent Thonny update (unsure which version but around Version 4.1.0) I have no longer been able to find any of our MicroPython packages using Tools > Manage packages.

These used to appear in older versions, and I have updated the **classifiers** and **keywords** in one of our packages to see if this helped, but I still can't find the package using the search.

It's a PyPI package and the search query I'm using is "Kitronik" with the package setup including:
``` python
...
setup(
    ...
    classifiers=[
        'Programming Language :: Python :: Implementation :: MicroPython',
    ],
    keywords='micropython, package',
    ...
)
```

Is there anything I'm missing that will allow our packages to appear in the Thonny package manage?

Thanks,
Jack

aivarannamaa, 2023-08-11:
Thank you for the report!

In MicroPython mode Thonny tries to combine results from PyPI and micropython.org and I tried to omit irrelevant CPython packages by lowering the required distance between package names and search terms, but I probably made it too strict. 

I'll look for another solution in Thonny 5.0, but in the meanwhile try searching for exact (or more precise) package name.

aivarannamaa, 2025-05-27:
PyPI search can't be used anymore. For less common packages the exact package name is the only option now.

closed 2025-05-27 by aivarannamaa