thonny/thonny #2886: esp32 Using this bytearray() function causes thonny to be very stuck

iBigDane, 2023-08-02:
https://github.com/micropython/micropython/issues/12140
Can you help me look at the above problem
I use micropython 1.19.1,use esp32-wrover-e whith 8MB SPIRAM.
I want to store the recording information of I2S in bytearray().
run the code
`buf = bytearray(512*1000)`
make my micropython become very slower,Even if I just print a letter,print("Hello world").
This problem does not occur with putty

aivarannamaa, 2023-08-05:
Do you have "Variables" and / or "Object inspector" pane open in Thonny?

iBigDane, 2023-08-05:
[image]
No panels are open

aivarannamaa, 2023-08-11:
Thanks!

It looks like Thonny's helper code is leaking memory. I don't know why it makes MicroPython slow, though. Will investigate later together with #1797.

MichaelBell, 2024-07-06:
I'm seeing this on an RP2040 too.  It looks like Thonny is somehow using all of the memory on the device after an allocation of a moderately large bytearray.

If I do this at the Thonny terminal:
```
MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> import gc
>>> gc.mem_free()
220896
>>> test1 = bytearray(64*1024)
>>> gc.mem_free()
.... <Long pause> ....
7376
```
nearly all the memory is consumed and the `gc.mem_free()` takes multiple seconds to run.

Whereas if I open a USB serial terminal to the same board running the same Micropython directly:
```
MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> import gc
>>> gc.mem_free()
228464
>>> test1=bytearray(64*1024)
>>> gc.mem_free()
162288
```
and `gc.mem_free()` completes immediately.

This Micropython issue suggests this is a common problem: https://github.com/micropython/micropython/issues/12325