thonny/thonny #2887: Zoom in with Ctrl+=

MattTheBanana, 2023-08-03:
Most websites use **Ctrl+=** on top of **Ctrl++** because it's uncomfortable to have to switch on and off pressing **Shift**. It would be a pretty easy implementation that'd be a nice QOL improvement.

aivarannamaa, 2023-08-11:
According to my test, Chrome uses Ctrl+= for restoring the zoom to 100%. Therefore I don't think it's a good shortcut for zooming in.

Moosems, 2023-08-11:
Command/Control += is used for zooming in and Command/Control _- is used for zooming out on Chrome for me.

Moosems, 2023-08-11:
He means the buttons above the brackets (`{[}]`)and `p`.

aivarannamaa, 2023-09-27:
Sorry for the late reply!

I thin I understand it now. Are you using the US keyboard layout?  I'm accustomed to EU layouts (EE and SWE in particular), which require Shift for `=` and don't require it for `+`.

closed 2023-09-27 by aivarannamaa