thonny/thonny #2888: Thonny Error I’ve never seen

KeeLimePi3, 2023-08-05:
I’m not big on coding or anything, but I’ve been trying to get some mods to work with a fan made game called clangen. Many of the mods and add-on’s require thonny to run the script to startup the game, but it’s been crashing and giving me these weird errors:
[image]
[image]
[image]

If anyone knows how to fix this, any guidance is very much appreciated!

aivarannamaa, 2023-08-05:
Does it work properly when you run it with "Run => Run current script in Terminal" ?

KeeLimePi3, 2023-08-05:
> Does it work properly when you run it with "Run => Run current script in Terminal" ?

It does not. I tried it and it gave me the same error.

aivarannamaa, 2023-08-11:
Does it run outside of Thonny? If not, then I suppose you need to search elsewhere.