thonny/thonny #2891: shell not working

jlaiosa, 2023-08-09:
I am trying to pip install a package on Thonny but the shell will not come up.  I am using a Mac 10.15.7 and Thonny 3.2.13.  I have been try to install adafruit_display_shapes using the Manage Package Tools.  It says it installed it but when I look into the library folder it is not there to I tried the shell but it does work.

Thanks for the help.

aivarannamaa, 2023-08-09:
Can you be more specific -- do you get an error message? Does nothing happen? Can you post a screenshot describing your problem?

jlaiosa, 2023-08-09:
I did eventually go the package to load using the manage plugs.  I can't show you a screen shot since it flashes away before I get it.  The only message I get is the following: Backend terminated or disconnected. Use 'Stop/Restart' to restart.

Thanks for replying.

jlaiosa, 2023-08-10:
Did you get my reply from yesterday?

aivarannamaa, 2023-08-10:
I did receive your reply. Unfortunately I still don't entirely understand your problem. Can you give me the steps for reproducing it?