thonny/thonny #2893: code completion hanging on suggestion selection       osx

harrih, 2023-08-10:
using current mac installation of Thonny on
 MacBook pro Intel   Big Sur   osx

 code completion hangs 
only keys to respond are Tab and Ctl + Space which just repeat same autocomplete
all others non responsive 

won't release until lose focus of the Thonny window and then return
 cursor then becomes available again

Thx   Harry

aivarannamaa, 2023-08-10:
Thank you for the report!

Does the completions list show up? Can you select an item with mouse click? Does it happen always? Which Thonny version are you using?

steeley, 2024-09-26:
I have been getting this bug for months  too. I'm on OSX 13.6.9. Thonny 4.14 and now on 4.16. autocomplete usually appears, so you tab to select the function  or whatever and when you try and fill in values or add the completion bracket its frozen. Makes what is a decent program almost unusable.