thonny/thonny #2895: CTRL + SPACE suggestions / autocomplete not working

MarioPL98, 2023-08-10:
[image]
In this example when I press CTRL and SPACE, nothing happens. I assume it should show possible methods or properties.

Thonny 4.1.1 with RP2040 board.

aivarannamaa, 2023-08-11:
Do you have Power Toys installed on this machine? If yes, then please see https://answers.microsoft.com/en-us/windows/forum/all/ctrl-space-not-working-while-pressed-together/f6ef4fae-5403-429a-8c8d-484095feee45

MarioPL98, 2023-08-11:
No, it's not installed. Ctrl-space works in pycharm, vscode, visual studio code, etc.

aivarannamaa, 2023-08-11:
I found the reason -- I had mostly tested in development environment, but the packaged variant of Thonny is missing one set of MicroPython typing stubs. 

Will fix this in Thonny 4.1.2.

aivarannamaa, 2023-08-11:
Please test https://github.com/thonny/thonny/releases/tag/v4.1.2

closed 2023-08-11 by aivarannamaa