thonny/thonny #2897: arm64 ubuntu based tarballs ?

capecrusader-121, 2023-08-13:
Currently only amd64/x86_64 ubuntu based tarballs are provided if possible it would be good if arm64 tarballs are also provided.

aivarannamaa, 2023-08-13:
It would be extra work for me. Actually I've considered getting rid of even the x64 binaries for Linux and rely only on the installation script, which creates a virtual environment with system Python 3.

capecrusader-121, 2023-08-13:
@aivarannamaa  I get it but killing the existing binaries(amd64) will indirectly kill the snap packages. 💀 😵 
As a single person it will be too much for me to build thonny from source and maintain it given also the complex and broken snap python plugins.

capecrusader-121, 2023-08-15:
@aivarannamaa  Should i create a PR For updating deps of this file as per 20.04 ?
-->  https://github.com/thonny/thonny/blob/master/packaging/linux/install_tools_ubuntu.sh

aivarannamaa, 2023-08-16:
The problem is that we (you and me) have different requirements for these binaries.

I introduced the binary Linux bundles so that people with older Linuxes could get recent versions of Python and Tk. For this to work, I needed to build the bundle in an old Linux, so that it doesn't require new versions of glibc for example.

You want the binaries to be a good fit for creating snaps for recent Linuxes (#2829). 

These is an inherent tension between these two requirements. Therefore I suggest you to start building binaries for snaps yourself. You could try my scripts or perhaps create a GitHub action for building the bundle. 

I may or may not continue with providing Linux bundles, but I surely don't want to publish two sets of them -- one for older distros and another for creating snaps for recent distros.

closed 2024-02-21 by capecrusader-121