thonny/thonny #2900: Idea for CTRL_C_EVENT on Windows

pan324, 2023-08-15:
I was curious if the following approach has been tested already. The idea is that we can run `windll.kernel32.SetConsoleCtrlHandler(0,0)` to enable CTRL_C_EVENT for all processes in a process group. This setting propagates to subprocesses as well. The remaining issue is that signals do not like to be sent across process borders very much so the frontend process has to attach to the newly created backend console (instead of connecting to a dummy). The approach has the following effects:

- If the frontend has a console attached, it has to give it up (in effect, nothing will be written in the cmd window). Might need to handle both cases separately so that GUI users can benefit while everything stays the same for console users.
- time.sleep can be interrupted just fine.
- It behaves like a proper ctrl+c in the interpreter, so the interrupted subprocesses (of subprocesses...) will show their own stderr instead of silently shutting down. This may be either totally annoying when running dozens of processes or quite helpful to see where a process was stuck.

Demonstration (make sure to run from IDE and not cmd due to the issue above):

```py
import os,sys,signal
import subprocess as sp
from threading import Thread
from collections import deque
from time import sleep
from ctypes import windll
os.environ["PYTHONUNBUFFERED"] = "1"

BACKEND = """print("Needed to initialize the (hidden) console on Windows.")
from time import sleep
from ctypes import windll
windll.kernel32.SetConsoleCtrlHandler(0,0)  # Enable CTRL_C_EVENT for this and all subprocesses.
sleep(2000)  # Only ctrl+c can stop this.
"""

p = sp.Popen(["python","-c",BACKEND],
             stdout=sp.PIPE, stderr=sp.PIPE, stdin=sp.PIPE,
             creationflags = sp.CREATE_NEW_PROCESS_GROUP|sp.CREATE_NO_WINDOW,
             encoding="u8")
p.stdout.readline()
windll.kernel32.FreeConsole()  # No harm for GUI, not optimal for console application.
windll.kernel32.AttachConsole(p.pid)

def Piper(stream, target):
    """Piper thread to funnel stdout and stderr into a deque of messages."""
    for line in stream:
        msgs.append((target, line))

# Start up the daemons and deque of msgs.
msgs = deque()
Thread(target=Piper, args=(p.stdout,sys.stdout), daemon=True).start()
Thread(target=Piper, args=(p.stderr,sys.stderr), daemon=True).start()

# Schedule an interrupt in 0.2 seconds to simulate a user requesting an interrupt.
def delayedinterrupt():
    sleep(0.2)
    os.kill(p.pid, signal.CTRL_C_EVENT)
Thread(target=delayedinterrupt).start()

# Print the messages that are being pushed by the daemons.
# Keep going as long as we either have messages or the process lives.
while msgs or p.poll() is None:
    try:
        target,msg = msgs.popleft()
        print(msg, end="", file=target)
    except IndexError:
        sleep(0.01)
print("ALL DONE")

```

aivarannamaa, 2023-08-17:
Thank you for these insights! I'll try to find time for experimenting with this at the weekend.

pan324, 2023-08-18:
There are even more subtleties. One issue with turning SIGBREAK into a KeyboardInterrupt is that the handler for whatever reason waits until sleep is done (well, the implementation of time.sleep is literally to wait for SIGINT with a timeout). In contrast, if there is no handler, then SIGBREAK shuts down everything ASAP. This includes all processes attached to the process group which most likely is the entire backend unless the user explicitly used subprocess with CREATE_NEW_PROCESS_GROUP (and in that case it's on them to take care of their processes).

So with `windll.kernel32.SetConsoleCtrlHandler(0,0)`  ctrl+c can be used as expected, AND ctrl+break (without touching the handler!) can be used to shut down the backend cleanly. 

Consider this line which start a new spinning process whenever running this code. `sp.Popen(["python","-c","while 1: pass"])`
Currently this will start up a new process with every F5 which will never shut down. By using ctrl+break for its intended purpose this issue will be fixed as a side effect. Haven't tested it yet but I think the same applies to multiprocessing (unless it does start a new process group I suppose).

aivarannamaa, 2023-09-27:
Thanks a lot for this information! I haven't had chance to experiment with this, but I'll do it before releasing Thonny 5.0.0.

BTW, Thonny 5 drops support for Windows 7, 8 and 8.1. Would this make the problem easier?