thonny/thonny #2901: In macOS, Command+N doesn't create a new file.

inventor7777, 2023-08-15:
When I hover my cursor over "New" it shows that the keyboard shortcut is Command + N. But when I hit Command+N, **nothing happens**.

aivarannamaa, 2023-08-17:
Thank you for the report! Unfortunately I can't reproduce this problem.

Which Thonny version and macOS version are you using? Do you have caps-lock on while experiencing this?

inventor7777, 2023-08-19:
Thank you for the reply! I’m using an M1 MacBook Air with both Ventura 13.5 and Sonoma beta 4 with a Logitech MX Keys keyboard for Mac. Caps lock wasn’t on. The open and save actions work though. Thonny is on 4.1.1, Python on 3.10.11, and it indicates that it’s on the ARM version.
Thank you! -Gabriel

aivarannamaa, 2023-09-27:
Sorry for the late reply!

Does the keyboard shortcut work if you use your laptop's keyboard instead of Logitech?

aivarannamaa, 2023-09-27:
That's strange... I can't reproduce this in my ARM Ventura. 

Which keyboard layout are you using?

inventor7777, 2023-10-01:
@aivarannamaa I'm using standard QWERTY layout.

closed 2024-05-31 by inventor7777