thonny/thonny #2903: toolbar icon too small

kjm1102, 2023-08-16:
Installed the latest thonny on  windows10 & the run, stop, etc icons in the tool bar along the top a tiny but fonts in other windows are OK. Can't figure out how to increase the size of just the icons?

aivarannamaa, 2023-08-17:
Thank you for the report!

Please try setting "UI scaling factor" to 2.0 or more at "Tools => Options => General".

If this works then please tell me which factor did you choose and what is your screen resolution.

kjm1102, 2023-08-17:
2 doesn't seem to make any difference to the icon size. 2.5 gives a reasonably sized icon but the font is now bigger than I'd like. 
[image]

aivarannamaa, 2023-08-17:
You can control editor font size separately in the "Editor" tab of the configuration dialog.

kjm1102, 2023-08-17:
I'm off topic now but is it possible to change the default indent to 2 spaces instead of 4?

aivarannamaa, 2023-08-17:
> I'm off topic now but is it possible to change the default indent to 2 spaces instead of 4?

Nope, see #1963

closed 2023-08-17 by aivarannamaa