thonny/thonny #2904: Change the wording for Ukraine fighting Russia?

nbourre, 2023-08-17:
Hello,

I am a computer science teacher in French Canada. I use Thonny for my introduction to programming course.

I would like to display the Thonny page on the school's website so that our students can download the application. However, the administration is hesitant to post your page due to the banner indicating support for Ukraine to fight Russia. The wording is too strong for them. Would it be possible to modify the sentence? For example, "Help Ukraine defend its territory" or something similar?

Don't worry, all of Canada is behind Ukraine.

Thank you.

aivarannamaa, 2023-08-17:
Ukraine is fighting Russian invasion, not simply fighting against Russia. That's important and that is also the wording on Thonny's homepage. Which part of it is too strong?

nbourre, 2023-08-17:
I'm totally with you on this, but I'm not the guy who decides...

It's just the term "fighting" which in french might be perceived as too violent. It's seems that the people who are in administration are not very "bilingual" and don't understand the nuance with this word.

Here are some examples

- "Thonny 4 stands in solidarity with Ukraine during its challenges against Russia."
- "Thonny 4 supports Ukraine during its times of adversity against Russia."
- "Thonny 4 is dedicated to the resilience and spirit of Ukraine amidst its situation against Russia."
- "Thonny 4 expresses its solidarity with Ukraine in light of its recent challenges involving Russia."
- "Thonny 4 is dedicated to Ukraine defending it's territory against the Russian invasion."

aivarannamaa, 2023-08-18:
Sorry, none of these express what I intend to express in my call.

Calling the current situation a "challenge" makes me sick. Your last proposal is close, but it is not enough -- Ukrainians, beside defending their territory, are also fighting a well known bully, for which I, as an Estonian, am eternally grateful, as Estonia has felt the fists of this bully too well. You'll understand If you learn our history and pay attention to Russian public rhetoric and leaked plans regarding the Baltic states before and after the start of the current war.

As I see it, in current situation the fight is essential and the feelings of a school administration are not.

closed 2023-08-18 by aivarannamaa