thonny/thonny #2905: 64-bit program

Denisov21, 2023-08-17:
HI,
I open this ticket because when I installed the 64-bit version of the program on my PC and I selected the "Install for all users" item. The program went to the folder **c:\program files (X86)\Thonny**, I would have expected the folder for 64 bit programs 
**c:\program files\Thonny**

aivarannamaa, 2023-08-18:
Thank you for the report!

I'll re-check later, but I may have misunderstood something in Inno Setup. Will re-read https://jrsoftware.org/ishelp/index.php?topic=32vs64bitinstalls

jonkloske, 2025-06-26:
Can confirm I also see this. Also, the binary thonny.exe has a compiled file version of 1.0.0.1 which makes it way more annoying to target for SCCM deployments. adjusting the rc file to match the software version number would definitely help here.