thonny/thonny #2906: Regarding Flashing Firmware in ESP32

shwetaguha, 2023-08-18:
I am getting an error as well as in the selection of Port in the dialog box getting this option: USB Serial @ COM3. I want to know if this is correct selection since I expected USB Serial CH340 COM3 and installed latest version of CH340 driver which is even showing under the Device Manager Port.

I tried re-installing driver CH340 as well as Reset push button of ESP32 board. However, I am still getting: USB Serial @ COM3.

Unable to flash the firmware on to ESP32

aivarannamaa, 2023-08-18:
If the port appears when you plug in your device and disappears when you unplug it, then it is the right port.

Can you be more specific about the error?

shwetaguha, 2023-08-19:
Hi Aivar,

Thanks for the input and at the moment I am able to run the python codes on
the ESP32 board.

Best Regards,
Shweta

closed 2023-08-19 by aivarannamaa