thonny/thonny #2908: Can't install package `micropython-bme280`

wyan, 2023-08-21:
Running on Ubuntu 23.04 on x86-64, installing package `micropython-bme280` fails with:

```
Looking in indexes: http://127.0.0.1:36628
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPConnectionPool(host='127.0.0.1', port=36628): Read timed out. (read timeout=15)")': /micropython-bme280/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPConnectionPool(host='127.0.0.1', port=36628): Read timed out. (read timeout=15)")': /micropython-bme280/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPConnectionPool(host='127.0.0.1', port=36628): Read timed out. (read timeout=15)")': /micropython-bme280/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPConnectionPool(host='127.0.0.1', port=36628): Read timed out. (read timeout=15)")': /micropython-bme280/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPConnectionPool(host='127.0.0.1', port=36628): Read timed out. (read timeout=15)")': /micropython-bme280/
ERROR: Could not find a version that satisfies the requirement micropython-bme280 (from versions: none)
ERROR: No matching distribution found for micropython-bme280
```

I have not checked installing other packages, and have no idea why it's looking for a local repository.

wyan, 2023-08-21:
Same problem with `micropython-ssd1306`

aivarannamaa, 2023-09-04:
Sorry for the late reply!

I can't reproduce this problem with the latest Thonny version 4.1.2, so I suggest to try it.

aivarannamaa, 2025-05-27:
Please reopen if the problem still exists with the latest Thonny