thonny/thonny #2909: NameError: Logger is not defined

AdrianSR7, 2023-08-22:
The IDE worked fine until... every time I press a key while editing a file stored on Raspberry Pi Pico. 

Traceback (most recent call last):
  File "C:\Users\Adrian\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\locals_marker.py", line 99, in update
    highlight_positions = self.get_positions()
  File "C:\Users\Adrian\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\locals_marker.py", line 66, in get_positions
    process_scope(child)
  File "C:\Users\Adrian\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\locals_marker.py", line 27, in process_scope
    process_node(child, local_names, global_names)
  File "C:\Users\Adrian\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\locals_marker.py", line 60, in process_node
    process_node(child, local_names, global_names)
  File "C:\Users\Adrian\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\locals_marker.py", line 60, in process_node
    process_node(child, local_names, global_names)
  File "C:\Users\Adrian\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\locals_marker.py", line 60, in process_node
    process_node(child, local_names, global_names)
  [Previous line repeated 4 more times]
  File "C:\Users\Adrian\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\locals_marker.py", line 44, in process_node
    if node.is_definition():  # local def
  File "C:\Users\Adrian\AppData\Local\Programs\Thonny\lib\site-packages\parso\python\tree.py", line 208, in is_definition
    return self.get_definition(include_setitem=include_setitem) is not None
  File "C:\Users\Adrian\AppData\Local\Programs\Thonny\lib\site-packages\parso\python\tree.py", line 235, in get_definition
    if self in node.get_defined_names(include_setitem):
  File "C:\Users\Adrian\AppData\Local\Programs\Thonny\lib\site-packages\parso\python\tree.py", line 1043, in get_defined_names
    return [
  File "C:\Users\Adrian\AppData\Local\Programs\Thonny\lib\site-packages\parso\python\tree.py", line 1047, in <listcomp>
    for name in _defined_names(self.children[i], include_setitem)
  File "C:\Users\Adrian\AppData\Local\Programs\Thonny\lib\site-packages\parso\python\tree.py", line 1011, in _defined_names
    names += _defined_names(child, include_setitem)
  File "C:\Users\Adrian\AppData\Local\Programs\Thonny\lib\site-packages\parso\python\tree.py", line 1017, in _defined_names
    if trailer.children[0] == '.':
AttributeError: 'Name' object has no attribute 'children'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Adrian\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1921, in __call__
    return self.func(*args)
  File "C:\Users\Adrian\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 839, in callit
    func(*args)
  File "C:\Users\Adrian\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\locals_marker.py", line 86, in perform_update
    self.update()
  File "C:\Users\Adrian\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\locals_marker.py", line 102, in update
    logger.exception("Problem when updating local variable tags")
NameError: name 'logger' is not defined

aivarannamaa, 2023-09-23:
Thank you for the report and sorry for the late reply! With which Thonny version did you get this error?

AdrianSR7, 2023-09-23:
Hi, the version was the latest available at that date, I don't know really
well which one is😅.

My best, Adrian.

aivarannamaa, 2023-09-25:
You can check Thonny version by opening the "Help" menu and selecting "About"

AdrianSR7, 2023-09-28:
Hi, sorry for the delay.

Versions are:
  - Thonny: 4.1.1
  - Windows 10
  - Python: 3.10.11
  - Tk: 8.6.13

Regards.