thonny/thonny #2911: GUI Layout - Window and Panels Not Retained After Quitting

farTooOld, 2023-08-23:
Re:  v4.1.2  (and, 4.0.2)
Mac 12.6.3 Monterey

Issue: After Quitting, Thonny does Not retain the Window/GUI layout (size of the various panels/Sidebars/etc).
I have to Resize/Re-Arrange it everytime (a PITA).

This has become my Favorite Python IDE - has ~almost everything needed without all the extra/clutter... THANK YOU!!!

aivarannamaa, 2023-08-24:
Thanks for the report!

I can't reproduce this in my mac, but I have plans to refactor Thonny's layout components in version 5.0 and most likely this will fix your issue.

farTooOld, 2023-09-02:
If it helps, attached images shows my arrangement and how Thonny looks after quitting and restart...
[image]
[image]

closed 2023-09-02 by farTooOld