thonny/thonny #2919: Thonny runs once on Kubuntu and then crashes on next tries

mcinsand, 2023-09-02:
Thonny has been running fine on the Windows 11 partition of my laptop, but it has been having issues on the Kubuntu partition.  I dug in yesterday and am not making headway, though I did find a tedious workaround.  If uninstalling and reinstalling, Thonny will run well until the application is closed.  On the second attempt to run Thonny, I get the error code below.  I saw the mention of pixel size, but I'm not finding the .thonny folder with the configuration file.  However, I did find 70-snap.thonny.rules under /etc/udev/rules.d folder.  Any thoughts?  

Regards

Traceback (most recent call last):
  File "/snap/thonny/142/lib/python3.10/site-packages/thonny/workbench.py", line 524, in _start_runner
    self._runner.start()
  File "/snap/thonny/142/lib/python3.10/site-packages/thonny/running.py", line 155, in start
    self.restart_backend(False, True)
  File "/snap/thonny/142/lib/python3.10/site-packages/thonny/running.py", line 730, in restart_backend
    self._proxy = backend_class(clean)
  File "/snap/thonny/142/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 91, in __init__
    self._fix_port()
  File "/snap/thonny/142/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 114, in _fix_port
    potential = self._detect_potential_ports()
  File "/snap/thonny/142/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 199, in _detect_potential_ports
    all_ports = list_serial_ports()
  File "/snap/thonny/142/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 989, in list_serial_ports
    _PORTS_CACHE = _list_serial_ports_uncached(skip_logging=skip_logging)
  File "/snap/thonny/142/lib/python3.10/site-packages/thonny/plugins/micropython/mp_front.py", line 1023, in _list_serial_ports_uncached
    for p in comports():
  File "/snap/thonny/142/lib/python3.10/site-packages/serial/tools/list_ports_linux.py", line 102, in comports
    for info in [SysFS(d) for d in devices]
  File "/snap/thonny/142/lib/python3.10/site-packages/serial/tools/list_ports_linux.py", line 102, in <listcomp>
    for info in [SysFS(d) for d in devices]
  File "/snap/thonny/142/lib/python3.10/site-packages/serial/tools/list_ports_linux.py", line 48, in __init__
    num_if = int(self.read_line(self.usb_device_path, 'bNumInterfaces'))
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'

capecrusader-121, 2023-09-03:
Are you trying to connect some usb device or raspberry pi like stuff ?
Try purging the snap & rm -rf ~/snap/thonny & then re-install .

Also report snap specific issues at : https://bugs.launchpad.net/thonny-snap/+filebug

Also to get better help do file this bug at the snap bug tracker.

aivarannamaa, 2023-11-03:
Similar to #2975