thonny/thonny #2921: quote key suddenly stopt working

wist51, 2023-09-05:
Hello,

I have a thonny ide working on my chromebook (Linux penguin) and for some reason my ' /¨ (quote/ double quote) - key stopt working in my thonny environment . Is this a known issue and/or can somebody help me with this problem?

regards wist51

aivarannamaa, 2023-09-05:
Do you mean you can enter neither quotes nor double-quotes in Thonny's editor? Does the key work in other programs?

wist51, 2023-09-05:
That is what I mean! and yes in other programs it works fine!

aivarannamaa, 2023-09-27:
Sorry for the late reply!

Do you know which Debian version your ChromeOS is using? What does `uname -a` Terminal command say?

Could you post your Thonny's "Help => About" content (e.g. as a screenshot)?

wist51, 2023-09-28:
Hope that the answer below helps:
 
***@***.***:~$ uname -a
Linux penguin 5.15.117-19680-g54e910f94133 #1 SMP PREEMPT Tue Sep 5 18:43:43 PDT 2023 x86_64 GNU/Linux
***@***.***:~$ ***@***.***:~$
 
***@***.***:~$ cat /etc/debian_version
11.6
 
Google Chrome:
Versie 116.0.5845.210 (Officiële build) (64-bits)
 
Regards
Willem Stoffelen

wist51, 2023-09-28:
> Hope that the answer below helps:
>
> Linux penguin 5.15.117-19680-g54e910f94133 #1 SMP PREEMPT Tue Sep 5 18:43:43 PDT 2023 x86_64 GNU/Linux
>
> 11.6
>
> Google Chrome:
> Versie 116.0.5845.210 (Officiële build) (64-bits)
>
> Regards
> Willem Stoffelen

Sorry, screenshot "Help => About" content forgotten! Herewith anyway.

aivarannamaa, 2023-09-28:
Unfortunately the pictures sent via e-mail don't show up in GitHub.

It suffices if you tell me Thonny version and Tk version as shown on the About box.

wist51, 2023-09-28:
The Thonny version is: Thonny 3.2. - Linux (64 bit) - Python 3.9.2 (64 bit) - Tk 8.6.11

aivarannamaa, 2023-09-28:
Thonny 3.2 is pretty old.

It may be the newest version offered in your package repository, though.

If you have x86_64 Chromebook, then you could `sudo apt remove thonny` and install Thonny+Python 3.10 bundle instead by executing `bash <(wget -O - https://thonny.org/installer-for-linux)`

wist51, 2023-10-13:
Dear Mr Aivar Annamaa,
 
After installing version 4.1.3 the 'quote/double quote' key still does not work on my chromebook but the rest of the environment works fine.
I do have to start the environment via the command line in the terminal and would like to have back the Thonny icon in my application laucher. Can you show me how to accomplish that? Thank you in advance.
 
Regards
Willem Stoffelen.

aivarannamaa, 2023-11-16:
The installer is supposed to create the shortcut in regular Linux distros, but unfortunately I don't know ChromeOS well enough to provide any guidance here.