thonny/thonny #2925: Garbled text in Shell, beeping sound, and rapid blinking onboard LED after flashing ESP8266 Wemo/Lolin D1 Mini with MicroPython

SimUKdev, 2023-09-08:
I'm using Thonny 4.1.2 on a Raspberry Pi 4 Model B 4GB with the Raspberry Pi OS.

When I try to update the MicroPython on an ESP8266 (Wemos/Lolin D1 Mini) it shows garbled text in the Shell on Thonny and a rapid beeping noise which I hear on the headphones plugged in to my Pi. The D1 Mini also flashes the onboard LED very rapidly, which continues to happen if pressing the onboard Reset button or even disconnecting and re-connecting the USB to the board.

If I press the "Stop/Restart" button in Thonny it'll refresh the Shell and stop the beeping sound, which seems to be generated by the Thonny Shell (as pressing the reset button on the D1 Mini when not using Thonny doesn't make the Pi 4 beep).

The only way I've been able to "fix" the D1 Mini is to manually flash the MicroPython firmware via esptool in the Pi's Terminal window. Once I've done that, Thonny can correctly recognise and communicate with the ESP8266 Lolin/Wemos D1 Mini again. But this seems a little confusing to me, as Thonny claims it's using esptool to flash the firmware anyway, so not sure what is happening differently.

What I have noticed is that if the D1 Mini is connected to Thonny and I press the onboard Reset button on the D1 Mini, it prints a shorter amount of the garbled text to the Shell in Thonny, and also makes the beep sound once. If I press the onboard Reset button a few times rapidly it prints more and more garbled garbage text in the Shell, with multiple beep sounds, one for each press of the reset button.

**Long description info about the issue:**

- Example of pressing the onboard Reset button on the ESP8266 Wemos/Lolin D1 Mini once:

```
>>> ;d��<ln;�ۛg��d`p�o�s�����c'�<d$��#��'g�g{���n�l�l8�'�{������c'�<d#��'g�l';�ۛg���drۓo��d����o�;��n<�d$dl #��|;�$�o��o�
MicroPython v1.20.0 on 2023-04-26; ESP module with ESP8266
Type "help()" for more information.
```

The garbled text is in a redish pink colour.

If the D1 Mini is connected via USB to Thonny, and I try to run esptool from the Pi's Terminal, I'll hear the beep sound a couple of times and some garbled garbage text in the Thonny Shell before it disconnects, and the Terminal establishes connection to the D1 Mini. The beeps only happen when Thonny has an open connection and not if the connection and not if the connection has been stopped.

- If I run esptool from the Terminal to get details of the connected board:

```
$ esptool.py flash_id
esptool.py v4.2.1
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: d8:f1:5b: (removed end of MAC address for this github issue)
Uploading stub...
Running stub...
Stub running...
Manufacturer: ef
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

When I've been flashing new MicroPython firmware to "fix" the D1 Mini (I've got several, they've all shown the same issue) I've been using the following command and had the following output:
```
$ esptool.py --port /dev/ttyUSB0 write_flash --flash_size=4MB -fm dio 0 ESP8266_GENERIC-20230426-v1.20.0.bin
esptool.py v4.2.1
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: d8:f1:5b: (removed end of MAC address for this github issue)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0009afff...
Flash params set to 0x0240
Compressed 634016 bytes to 420365...
Wrote 634016 bytes (420365 compressed) at 0x00000000 in 37.5 seconds (effective 135.2 kbit/s)...
Hash of data verified.
[... 137 lines cut ...]
esptool.py v4.2.1
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: d8:f1:5b: (removed end of MAC address for this github issue)
Uploading stub...
Running stub...
Stub running...
Manufacturer: ef
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

And again the "fix" is to use esptool in the Pi's Terminal to re-flash the MicroPython firmware `esptool.py --port /dev/ttyUSB0 write_flash --flash_size=4MB -fm dio 0 ESP8266_GENERIC-20230426-v1.20.0.bin`

I've got genuine Wemos/Lolin D1 Minis, I've tried both v3.1.0 and v4.0.0 boards (only real difference between them is the USB connector from Micro USB to USB Type-C). I can install the updated Micropython via esptool, so that is a workaround, but something doesn't seem right about how it's flashing/updating the Micropython firmware via Thonny.

I have seen from looking on Google multiple reports over the years of the ESP8266 having a different baud rate for the serial when it boots, so possibly this is partly an issue, and also reports of the reference point for the flash size causing issues which can result in the ESP8266 getting stuck in a reboot loop. I note that esptool reports `Detected flash size: 4MB` yet  "Show image info" in Thonny shows `Flash size: 512KB`, again i don't know if that's part of the issue, but it stood out when comparing the data.

SimUKdev, 2023-09-08:
Some of the results I found when looking at similar reports of issues of the serial monitor garbage/blinking onboard LED:

https://spsarolkar.github.io/esp8266/2018/03/18/Blinking-blue-LED-of-death-ESP8266.html

https://github.com/esp8266/Arduino/issues/4005

https://community.platformio.org/t/serial-monitor-goes-havoc-when-spinning-up-d1-mini-with-a-load-on-5v/20297

https://forum.micropython.org/viewtopic.php?t=8915

https://docs.espressif.com/projects/esptool/en/latest/esp8266/esptool/serial-connection.html

SimUKdev, 2023-09-08:
I've just found that the sound I'm hearing is the "notification sound" that can be switched off in Thonny `Tools > Options > General > Disable notification sound`, so that confirms it's coming from Thonny, but doing that in itself doesn't solve the overall issue that is occurring after MicroPython has been flashed via Thonny, or the garbled text on a reset any other time.

SimUKdev, 2023-09-09:
#2801 has lead me to find the "Show install options" which I'd overlooked before, and gives five more settings (Target address, Install Speed, Flash mode, Flash size, Disable stub loader), and changing the "Flash size" from "keep (read the setting from the selected MicroPython image)" to "detect (uses detection based on flash ID)" has allowed Thonny to correctly flash MicroPython 1.20 to the board.

Would it be better (more compatible/less chance of install errors?) for the default on the `Flash size` option to be `detect` instead of the size specified by the MicroPython image?

I'm still finding the garbled text in the Shell on each press of the onboard Reset button, so still not 100% sure the board is working correctly, but at least have a D1 Mini which appears to be functioning as I'd see from a manual esptool installation.

closed 2023-09-13 by SimUKdev

aivarannamaa, 2023-09-14:
Thank you for the report and sorry for the late reply!

> Would it be better (more compatible/less chance of install errors?) for the default on the Flash size option to be detect instead of the size specified by the MicroPython image?

This would be unexpected for at least some users, because `keep` is the default in `esptool` (https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/flash-modes.html#flash-size-flash-size-fs)