thonny/thonny #2929: Thonny Prevents pygame Window Close

troywein, 2023-09-11:
Hello,

When using pygame in combination with the Thonny IDE on Mac OS, calling pygame.quit() doesn't close the open window.  If running outside Thonny the window closes as expected.  The attached file is a short example that can be used to reproduce this issue on the Mac.
____________
[window_hang.py.zip](https://github.com/thonny/thonny/files/12579174/window_hang.py.zip)

aivarannamaa, 2023-09-11:
This is expected. Thonny doesn't close the process after the script ends in order to allow inspecting the resulting variables. 

You can force closing the process by adding `exit()` after the `pygame.quit()`.

troywein, 2023-09-11:
> Caution!
> **Don’t click on links or attachments unless you trust the sender. This email came from an external source.**

Hello Aivar,

Thanks for the quick response.  The exit() isn’t an option for our course as we would like the cadets to do some additional work after the window closes.  Additionally, it closes as expected in Thonny on Windows so is causing some confusion in our course between the Windows and Mac users.  Thanks Troy

Dr. Troy Weingart
Professor of Computer & Cyber Sciences
Office: Fairchild Hall, Suite 6G-115
DSN: 333-7658 COMM: 719-333-7658

aivarannamaa, 2023-09-23:
Sorry for the late reply!

> The exit() isn’t an option for our course as we would like the cadets to do some additional work after the window closes.

Have you been able to achieve the desired behaviour on macOS in another IDE or command line?

According to my test, on macOS the Pygame windows remains hanging even outside of Thonny, e.g. by doing `python3 -i window_freeze.py` (please note the `-i` switch, which forbids the process to close after the script ends).

Perhaps this issue should be discussed in Pygame's issue tracker?

troywein, 2023-09-23:
Hello,

Yes, Pygame works with Visual Studio Code and the window closes as expected.  This script also works as expected via the command line (it doesn’t hang and closes appropriately).  The script will run to completion in Thonny on MacOS with no errors reported, however you must click the stop sign in the menu bar to close the window.  This behavior has been duplicated by many students and and instructors.

There is something about Thonny on MacOS vs Windows vs command line that is preventing the Pygame window from closing.  Thank you for investigating this.

aivarannamaa, 2023-09-27:
Just to be clear -- have you been able to close Pygame window without problems AND continue using the same process via Python REPL in macOS?

aivarannamaa, 2023-09-27:
> When using Python REPL (interactive) the window does not close as it does when running the same code as a script on MacOS.

That's the reason Pygame window doesn't close in Thonny and it's Pygame's issue. 

Thonny's work-around is to use "Run => Run current script in terminal".

closed 2023-09-27 by aivarannamaa

troywein, 2023-09-27:
Thanks for your work on this, however, I don’t understand why it works via the terminal and also why it works just fine in Thonny on Windows (btw, also works fine in VSC on Mac and PyCharm on Mac).  Do you have any advice to give to the pygame team?  I’m guessing they will point to Thonny as the issue being that it works just fine in other IDE’s and also via the terminal.

aivarannamaa, 2023-09-27:
I thought you just confirmed yourself that Pygame doesn't close normally even outside of Thonny on macOS:

> When using Python REPL (interactive) the window does not close as it does when running the same code as a script on MacOS.

troywein, 2023-09-27:
Only when running in REPL.  It closes fine when running as a script.

aivarannamaa, 2023-09-27:
> Only when running in REPL.  It closes fine when running as a script.

Well, that's what I mean -- Thonny always runs the scripts in REPL.