thonny/thonny #2932: PROBLEM IN THONNY'S BACK-END: Internal error

AppsByDavideV, 2023-09-18:
Hello,
Problem as per title with Thonny 4.1.0 up to 4.1.2.

Here is the app console output that appears as soon I open Thonny:
`Traceback (most recent call last):
  File "main.py", line 405, in <module>
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
KeyboardInterrupt: 
MicroPython v1.20.0 on 2023-04-26; ESP32 module (spiram) with ESP32
Type "help()" for more information.
PROBLEM IN THONNY'S BACK-END: Internal error (thonny.plugins.micropython.mp_back.ManagementError: Script produced errors).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0.`

Attached the log.
[backend.log](https://github.com/thonny/thonny/files/12647088/backend.log)

PC: Win10 and Win11 no matter
Board ESP32 with SPI ram , Micropython 1.20
Thonny (current) 4.1.2

It looks like it is related to uasyncio. Using the Stop button just restart the board and I can see some log printed from boot.py then then the error above.

Note that with some other IDE I can work on the board, just a Thonny problem, but maybe it will be useful for others as well.
Thanks for your work!
\Davide

AppsByDavideV, 2023-09-24:
> Thank you for the report!
> 
> This is somewhat similar to #2814, which also describes a case where `_thonny_helper` (a helper object defined by Thonny) seems lack some methods that should be there.
> 
> Can you provide a small example program for reproducing this error?

Hello,
of course yes, I can.
As there are only few lines, I attach here the screenshots.
Just to recap: 
Device is ESP32 - mpy 1.20 latest
I started with a new fresh install of micropython, to be sure, and it connects fine. Only boot.py is present on board and it is empty (with some commented lines) 
I create main.py and it only creates a new directory in the chip memory, called 'sys'
Soft reboot the board -> can't connect anymore.

Using a different IDE: it says the chip is busy -> press stop and I can work on it again.
Hope this helps!
[image]

[image]

[image]
[image]

\Davide

AppsByDavideV, 2023-09-24:
To make it short, this is enough to block Thonny (via command line):
```
>>>import os
>>>os.mkdir('sys'
>>>
```
then reboot and it will not connect anymore . 
It could be related to the chip filesystem, Thonny or both.

To be able to use it again, the flash must be formatted:
```
# ESP8266 and ESP32
import os
os.umount('/')
os.VfsLfs2.mkfs(bdev)
os.mount(bdev, '/')
```

aivarannamaa, 2023-09-25:
Thanks for the investigation!

I now understand the situation. By creating a directory named `sys`, you are effectively shadowing the built-in module `sys`, which Thonny is relying on. Note that you wouldn't be able to use the original `sys` module in your own code either.

I don't know any means Thonny could overcome this situation. For any other module it could temporarily tweak `sys.path` and un-hide the original, but after hiding `sys` it is not possible, because the original `sys` is ... hidden.

AppsByDavideV, 2023-09-25:
Sorry, I apologize for that. Of course use of reserved keywords in code must be avoided. It was a port from a C++ code that uses a dir called 'sys'.
I formatted the flash, changed the name to 'syst' and it works great!
Thanks!
\Davide

aivarannamaa, 2023-09-27:
I'll keep this issue open for now and look for ways to make similar errors more explanatory.